notro/fbtft #492: Monochrome (1bpp) Support for ST7565 glcd

mayurpatelgec, 2017-12-10:
Hi @notro,

I am working on ST7565 based monochrome GLCD and i followed your and @zanaster work [zanaster/st7565](https://github.com/zanaster/fbtft/commit/3fd31cdcba9021575cd911ba329b9c5904ac726f) his [fb_st7565 FB driver.](http://zanaster.blogspot.com/2013/10/linux-driver-for-st7565.html) 
All goes well and initializes and shows initialization text. but when i try anything else then all pixels (99%) goes on. I came to know it is developed for 16 bpp and I need 1 bpp. Can you help me in it ?

notro, 2017-12-12:
How are you driving it from userspace?
Remember that any RGB565 value except pure black (zero) will turn on the pixel:
 ```vmem16[(y*8+i)*132+x] ? 1 : 0```

Sidenote:
This driver does grayscale conversion and dithering: http://elixir.free-electrons.com/linux/latest/source/drivers/staging/fbtft/fb_agm1264k-fl.c
But it's better to do this in userspace.

mayurpatelgec, 2017-12-12:
hi Thanks for reply..

actually after loading the driver with **_modprobe_** i want to use it for QT project but when i start the app with 

`./Helloworld -qws -Display LinuxFb:/dev/fb0` then display shows all the pixels. i also know that i have to create a project that uses only monochrome colors..

**when it's not possible to drive this display with QT then i want to know what are the other possible ways to drive this display from user space...**

notro, 2017-12-12:
I'm sorry I don't know. I haven't got any experience with graphics libraries.
A search for 'qt monochrome' gave me this: https://machinekoder.com/rendering-to-a-monochrome-oled-display-with-qml/

mayurpatelgec, 2017-12-13:
thanx. i am already working with it (compatibility issues Qt4 and Qt5 in QML)

can you suggest me any other option to drive this monochrome LCD with this fb driver?

notro, 2017-12-13:
As a said, I haven't got any experience with graphics libraries.

mayurpatelgec, 2017-12-13:
ok thank you for your support.. it was great help from your side.. i will keep updating if i got any success :)

closed 2021-08-24 by github-actions[bot]