notro/fbtft #493: st7789v can not display picture at full screen and the color is also wrong

SearchTheTruth, 2017-12-13:
Because I work at the kernel 3.4. So I use the old version of fbtft.
commit number is 06f0bba152c036455ae76d26e612ff0e70a83a82.
But i find that when I use Qt to display a picture ,the picture is very small at the central of the screen. And the color is also wrong. 
I use  spi to control, 16bpp  and width,height is 240, 320.
I just add st7789v in fbtft_device.c and dont  change other code.I cant find the reason, wish you could help me,thank you very much.

SearchTheTruth, 2017-12-15:
I find the reason。
The color is wrong because need to inversion the color.And can't full screen becaus qt environment need to modify.

closed 2017-12-27 by SearchTheTruth

wdmomoxx, 2018-02-07:
@SearchTheTruth Hi, I had the same problem. I don't know how to inversion the color. Please help me

SearchTheTruth, 2018-03-15:
I'm sorry, I just see you email right now.
Did you fix this problem?