notro/fbtft #495: Always white Adafruit 2.2" display

geizio, 2017-12-24:
Greetings, notro

I have this [ILI9340 2.2" diagonal LCD TFT display from Adafruit](https://www.adafruit.com/product/1480). I have wired it to my [Raspberry Pi Zero W](https://www.raspberrypi.org/products/raspberry-pi-zero-w/) according to this [](http://www.sudomod.com/forum/viewtopic.php?t=2312) forum with some help from [](https://learn.adafruit.com/user-space-spi-tft-python-library-ili9341-2-8/wiring). This is my first time soldering. I used a multimeter to verify that none of the connections were touching each other.

I am using the latest version of Raspbian Stretch as of today, Saturday, December 23 2017. I have followed a countless number of guides for setting the display up including [yours](https://github.com/notro/fbtft/wiki). The most useful so far has been:

> sudo apt-get update && sudo apt-get upgrade
> 
> # -- SCREEN --
> 
> sudo wget https://raw.github.com/Hexxeh/rpi-update/master/rpi-update -O /usr/bin/rpi-update && sudo chmod +x /usr/bin/rpi-update
> sudo rpi-update
> sudo shutdown -r now
> 
> # expand
> sudo raspi-config
> # remove or comment out the spi blacklist line
> sudo nano /etc/modprobe.d/raspi-blacklist.conf
> 
> sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
> sudo shutdown -r now
> 
> sudo modprobe fbtft dma
> sudo modprobe fbtft_device name=adafruit22a
> 
> # run now
> FRAMEBUFFER=/dev/fb1 startx
> con2fbmap 1 1
> 
> # ----- permanent -----
> 
> # add to file /etc/modules
> fbtft_device name=adafruit22a verbose=0
> 
> # Auto login: modify file /etc/inittab
> #1:2345:respawn:/sbin/getty --noclear 38400 tty1
> 1:2345:respawn:/bin/login -f pi tty1 </dev/tty1 >/dev/tty1 2>&1
> 
> sudo apt-get install xserver-xorg-video-fbdev
> Create file: /usr/share/X11/xorg.conf.d/99-fbdev.conf
> 
> Section "Device"  
>   Identifier "myfb"
>   Driver "fbdev"
>   Option "fbdev" "/dev/fb1"
>   Option "Rotate" "CCW"
> EndSection
> 
> # Auto startx: modify file /etc/rc.local
> #su -l pi -c "env FRAMEBUFFER=/dev/fb1 startx &"
> su -l pi -c "env con2fbmap 1 1 && startx &"
> 
> # Add kernel argument to file /boot/cmdline.txt 
> # Possibly map:11?
> #fbcon=map:10
> fbcon=rotate=3

from [](https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=66968). When I run `FRAMEBUFFER=/dev/fb1 startx`, my HDMI screen goes black accept for the terminal and my tft display does not change (it stays white). when I run `con2fbmap 1 1`, my HDMI screen does not change and it stays fully functional while my tft display does not change either (it stays white). The tft display always stays completely white accept for under two conditions:
1. When booting, because /etc/modules-load.d/fbtft.conf is loading fbtft_device kernel module from /etc/modprobe.d/fbtft.conf (I think), the tft display's backlight turns off or goes black because it is refreshing (I think).
2. When my HDMI screen goes black (still on) from being "AFK", the tft display's backlight turns off or goes black.
When my Raspberry Pi Zero W is plugged in (whether it has an SD card or not), the tft display's backlight turns white.

I have fb1, fb0, and the spidevs located in /dev folder. From all of this and the dmesg fbtft I really think that the tft display should already be working.

Thanks for your attention. I’m looking forward to your reply.

geizio, 2018-01-01:
Greetings, notro

I have now tried to use fbtft_device with a completely different raspberry pi (the first model instead of my [zero w](https://www.raspberrypi.org/products/raspberry-pi-zero-w/)) and a different display (a [2.2" ili9341 tft](https://www.amazon.com/4-Wire-Display-Module-240x320-ILI9340C/dp/B01N3L01BB/ref=sr_1_2?ie=UTF8&qid=1514778790&sr=8-2&keywords=ili9341+2.2) instead of my [2.2" ili9340c tft](https://www.adafruit.com/product/1480)). The [adafruit ili9341 library](https://github.com/adafruit/Adafruit_Python_ILI9341), `con2fbmap 1 1`, and `FRAMEBUFFER=/dev/fb1 startx` did not work on either of them. Still no luck.

Once again, thanks for your attention. I’m looking forward to your reply.

notro, 2018-01-05:
A white display usually comes down to either miswiring (or faulty) or a wrong initialization sequence.
If everything is working it should come up as a backlit black screen (all pixels are zero).
Running this should give you a console on the tft:
```
$ sudo modprobe fbtft_device name=adafruit22a
# check that it was registered
$ cat /sys/bus/spi/devices/spi0.0/modalias
fb_ili9340 or something
$ dmesg
there should be a fb_ili9340 registration message

# this should give a console
$ con2fbmap 1 1
```

geizio, 2018-01-05:
Thank you. I tried `con2fbmap 1 1` on both displays and dmesg said my displays were being registered but nothing happened. I have decided to try something else now.

Thanks again

closed 2018-01-05 by geizio