notro/fbtft #500: LGDP4524 supported ?

craftcore, 2018-02-05:
i am struggling to get the FB driver working with this type of display and raspberry pi
[https://www.aliexpress.com/item/2-0-3-3V-TFT-LCD-Touch-Screen-Breakout-Board-With-Touch-Pen-For-Arduino/32780151979.html] 
i can read the marking "driver IC LGDP4524"
which should be somehow compatible to ili9325
except it only supports 176x220 resolution
can you tell me if this driver can support it ?
i tried with custom device command `modprobe fbtft_device custom txbuflen=32768 name=fb_ili9325 buswidth=8 width=176 height=220 gpios=reset:4,dc:18,wr:17,cs:27,db00:23,db01:24,db02:10,db03:9,db04:25,db05:11,db06:8,db07:7`
wired it right
if i test it with `con2fbmap 1 1`
screen stays blank

notro, 2018-02-26:
It is likely that you need to overrride the intialization sequence (register setting). See if you can find some example code for the display.
Driver source code with default register settings: https://elixir.bootlin.com/linux/latest/source/drivers/staging/fbtft/fb_ili9325.c
How to change using init= argument: https://github.com/notro/fbtft/wiki/fbtft_device

jpbbarros, 2020-05-29:
> i am struggling to get the FB driver working with this type of display and raspberry pi [...]

Hello craftcore,
managed to connect your display to the raspberry?
I have an equal and needed information on how to connect (wiring scheme) and sample code in python.
Could I get some information?
Thank you,
Best regards,
João Barros

closed 2021-08-24 by github-actions[bot]