notro/fbtft #504: FBTFT driver -- OLED display (Monochrome) -- Write a single pixel or some pixels

Balajinp, 2018-02-06:
Trying to bring up the SSD1306 OLED display (Monochrome). Need to confirm that my device tree values are proper and the device spi access proper before display anything... 

Do you have the function to write the single pixel in OLED display? This function may be in kernel driver or user space applications? 

-Thanks.

Balajinp, 2018-02-16:
In write_vmem() in fb_ssd1306.c file, changed this

set the txbuf.buf to 0xFF instead of the user space values, this is for testing

ret = par->fbtftops.write(par, par->txbuf.buf,
                                  par->info->var.xres * par->info->var.yres /
                                  8);
then display prints full pixels on.

arunkvnss, 2019-11-11:
Do you have any answer for this ?

closed 2021-08-24 by github-actions[bot]