notro/fbtft #513: Yocto-based Raspberry Pi Image, ads7846_device command does not work

mozcelikors, 2018-03-24:
Hi,
I am trying to install this on a Raspberry Pi image with Yocto. I have successfully configured my custom ili9341 device so far. However, I have a problem getting ADS7846/XPT2046 touch controller to work.

The simple problem I'm having right now is:

`modprobe: module ads7846_device not found in modules.dep`

I have compiled ads7846.dtbo and put it in /boot/overlays in my image. I've compiled https://raw.githubusercontent.com/raspberrypi/linux/rpi-3.18.y/arch/arm/boot/dts/overlays/ads7846-overlay.dts this file using:

```
./dtc -I dts -O dtb -o ~/ads7846.dtbo ~/ads7846-overlay.dts
cp ~/ads7846.dtbo <...>/overlays
```

In config.txt I have,
```
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=ads7846,cs=1, penirq=17, speed=500000, swapxy, pmax=255, xohms=100, xmin=200, xmax=3900, ymin=200, ymax=3900
```

I am planning to start my device with following:
```
modprobe fbtft_device custom name=fb_ili9341 gpios=reset:25,dc:24,led:18 speed=16000000 rotate=270 bgr=1
modprobe ads7846_device pressure_max=255 y_min=190 swap_xy=1 y_max=3850 gpio_pendown=17 x_max=3850 x_min=230 x_plate_ohms=60
modprobe ads7846
FRAMEBUFFER=/dev/fb1 startx
```
All commands except ads7846_device works, So, the created overlay did not work for me. Am I missing something?
Also, when I tried

`modprobe ads7846 pressure_max=255 y_min=190 swap_xy=1 y_max=3850 gpio_pendown=17 x_max=3850 x_min=230 x_plate_ohms=60`

I get the following in my driver messages (dmesg):

```
[  150.318152] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[  150.330864] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[  150.333149] spidev spi0.0: spidev spi0.0 125000kHz 8 bits mode=0x00
[  150.333185] spidev spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
[  150.333260] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[  150.333321] spidev spi0.0: Deleting spi0.0
[  150.334381] fbtft_device: GPIOS used by 'fb_ili9341':
[  150.334407] fbtft_device: 'reset' = GPIO25
[  150.334418] fbtft_device: 'dc' = GPIO24
[  150.334427] fbtft_device: 'led' = GPIO18
[  150.334449] spidev spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
[  150.334468] spi spi0.0: fb_ili9341 spi0.0 16000kHz 8 bits mode=0x00
[  150.362614] fb_ili9341: module is from the staging directory, the quality is unknown, you have been warned.
[  150.764291] graphics fb1: fb_ili9341 frame buffer, 320x240, 150 KiB video memory, 16 KiB buffer memory, fps=20, spi0.0 at 16 MHz
[  362.039783] ads7846: unknown parameter 'pressure_max' ignored
[  362.039821] ads7846: unknown parameter 'y_min' ignored
[  362.039835] ads7846: unknown parameter 'swap_xy' ignored
[  362.039846] ads7846: unknown parameter 'y_max' ignored
[  362.039859] ads7846: unknown parameter 'gpio_pendown' ignored
[  362.039869] ads7846: unknown parameter 'x_max' ignored
[  362.039880] ads7846: unknown parameter 'x_min' ignored
[  362.039893] ads7846: unknown parameter 'x_plate_ohms' ignored
```

I think this thread would help people who also try to get this controller to work with a Yocto-based image. Any help is appreciated.

notro, 2018-03-24:
You don't neeed ads7846_device when you use a Device Tree overlay. They both add a device.
The dtoverlay directive should not have spaces in it, maybe that's your problem. This command should give you the bootloader DT log: ```sudo vcdbg log msg```, at least on stock Raspbian.

mozcelikors, 2018-03-24:
Your solution helped by the way, just noticing it. I have some other problems apparently. Thanks.

closed 2021-08-24 by github-actions[bot]