notro/fbtft #515: ST7789V not working

devikaesd, 2018-03-30:
I'm using ST7789V display with 4.9.41 Kernel. My DTS file looks like this.

&spi1 
{
     status = "okay";
     pinctrl-names = "default";
     pinctrl-0 = <&myspi1_pins_default>;
     ti,spi-num-cs = <2>;
     ti,pindir-d0-out-d1-in = <1>;
     #address-cells = <1>;
     #size-cells = <0>;   

         spidev@0
        {

          compatible = "sitronix,st7789v";
          buswidth = <0x8>;
          //spi-3wire = <0x01>;
          spi-max-frequency = <2800000>;
          reg = <0x481a0000 0x400>;
          dc-gpios = <&gpio0 7 0>;
          reset-gpios = <&gpio3 8 0>;   
                };
};
All pins looks fine, I'm able to see data, clk, cs, dc pins but not reset pins. And LCD only blinks when I'm sending some data.

I have configured for 8 bits 3-wire. For 3-wire I have shorted MOSI and MISO pins with 10K.

Any suggestion, what I'm missing in DTS file to make reset work and make my display up.

notro, 2018-04-08:
> reg = <0x481a0000 0x400>;

This should be the chip select number.

Is your spi device registered?
```
pi@pi2835:~$ ls -l /sys/bus/spi/devices/
total 0
lrwxrwxrwx 1 root root 0 Apr  6 15:15 spi0.0 -> ../../../devices/platform/soc/20204000.spi/spi_master/spi0/spi0.0
lrwxrwxrwx 1 root root 0 Apr  6 15:15 spi0.1 -> ../../../devices/platform/soc/20204000.spi/spi_master/spi0/spi0.1

```
This will show if the driver is loaded:
```dmesg | grep 'frame buffer'```

waelabd, 2020-06-08:
@devikaesd have you resolved this issue?

devikaesd, 2020-06-09:
Yes..it got solved..

I configured regular 4Wire SPI for that.

But in device tree you need to specify one pin and one pin as out.

Your's Sincere

temoto, 2020-07-11:
@devikaesd please please share final working dts.

wamanprabhu, 2020-07-13:
myspi1_pins_default: myspi1_pins_default {
	pinctrl-single,pins = <
		0x190 ( PIN_INPUT | MUX_MODE3 ) /* (A13) mcasp0_aclkx.spi1_sclk */
		0x194 ( PIN_INPUT | MUX_MODE3 ) /* (B13) mcasp0_fsx.spi1_d0 */
		0x198 ( PIN_INPUT | MUX_MODE3 ) /* (D12) mcasp0_axr0.spi1_d1 */
                /*Below line was modified from */
                //0x164 ( PIN_INPUT | MUX_MODE2 ) /* (C18) eCAP0_in_PWM0_out.spi1_cs1 */
                //to
		0x19c ( PIN_INPUT | MUX_MODE3 ) /* (C12) mcasp0_ahclkr.spi1_cs0 */
	>;
&spi1 {
	 status = "okay";
         pinctrl-names = "default";
         pinctrl-0 = <&myspi1_pins_default>;
	 ti,spi-num-cs = <2>;
	 ti,pindir-d0-out-d1-in = <1>;
	 #address-cells = <1>;
	 #size-cells = <0>;	  
	
         spidev@0{
                  compatible = "sitronix,st7789v";
		  buswidth = <0x8>;
		  //spi-3wire = <0x01>;
		  spi-max-frequency = <12000000>;
		  reg = <0x481a0000 0x400>;
		  reset-gpios = <&gpio3 8 0>;
		  dc-gpios = <&gpio0 7 0>;
                };
};

Hope this help

temoto, 2020-07-13:
FYI for ST7789 to display colors correctly, driver must contain this patch, merged into Linux since 5.8: https://github.com/torvalds/linux/commit/f03c9b7884720973d1673fbb64f808897ca88a12
Armbian has just accepted to apply this patch when building any kernel version.

Here's our working dts file for Orange Pi Lite:
```
/dts-v1/;
/plugin/;

/ {
        compatible = "allwinner,sun8i-h3";

        fragment@0 {
                target = <&spi0>;
                __overlay__ {
                        #address-cells = <1>;
                        #size-cells = <0>;
                        pinctrl-names = "default", "default";
                        pinctrl-1 = <&spi0_cs1>;
                        cs-gpios = <0>, <&pio 0 20 0>;

// NOTE irrelevant for display config, but we use spidev concurrently with chip select and it works
                        spidev@0 {
                                compatible = "spidev";
                                reg = <0>;
                                status = "okay";
                                spi-max-frequency = <16000000>;
                        };

                        st7789v@1 {
                                compatible = "sitronix,st7789v";
                                reg = <1>; // NOTE SPI chip select number
                                pinctrl-1 = <&st7789v_pins>;
                                spi-max-frequency = <32000000>;
                                spi-cpol;
                                spi-cpha;
                                rotate = <90>; // NOTE 90/270 for horizontal orientation
                                fps = <25>;
                                buswidth = <8>;
				txbuflen = <16384>; // NOTE supposed to speed up overall transfer and make SPI device more available for other uses
                                // NOTE obviously your pins will vary but last 0 or GPIO_ACTIVE_HIGH is important
                                reset-gpios = <&pio 0 10 0>;
                                dc-gpios = <&pio 0 9 0>;
                                led-gpios = <&pio 0 8 0>;
                                debug = <0>; // NOTE high value like 9 will print lots of debug info to dmesg
                        };
                };
        };

        fragment@1 {
                target = <&pio>;
                __overlay__ {
                        spi0_cs1: spi0_cs1 {
                                pins = "PA20";
                                function = "gpio_out";
                                output-high;
                        };

                        st7789v_pins: st7789v_pins {
                                pins = "PA8", "PA9", "PA10";
                                function = "gpio_out", "gpio_out", "gpio_out";
                        };
                };
        };

};
```

@notro thank you for your hard work. Have a great time.

closed 2021-08-24 by github-actions[bot]