notro/fbtft #518: need for help: framebuffer number assigned differently

khalifima, 2018-05-03:
I am using fbtft to display on five ili9340 compatible displays simultaneously using raspberry pi zero w 
i created the dt overlays and so far everything is working i could display on each of the displays. but the framebuffer numbering is somehow not deterministic  sometimes it assigns dev/fb1 to the display attached to spi0.1 other times to another display e.g spi1.0 . could someone tell whether it is possible to fix the framebuffer assignement to a specified spi bus number??

notro, 2018-05-10:
Since you use the same driver for all displays, it's the device order that decides framebuffer numbering.
Are the displays on the same spi bus with five chip selects, or on multiple busses?

tabarnakos, 2019-09-24:
Hey @khalifima,

I know this thread is 1 year old, but I can't for the life of me figure out how to map a single display from spi1.0 to fb0

Any chance you can share your code?

I just want to use the tft as some kind of debug console, but somehow every library I find will not work on fb1

notro, 2019-09-24:
The fbdev numbering (fbN) is controlled by the order in which the drivers are registered.
On the Pi, the firmware sets up a framebuffer that the early simplefb driver hooks into and it gets fb0:
```
[    1.043103] simple-framebuffer 1e887000.framebuffer: fb0: simplefb registered!
```
fb0 is what the console is attached to by default. See here for how to change that: https://github.com/notro/fbtft/wiki/Boot-console
You can also change it at runtime by using ```con2fbmap```.

In the case of the Pi, later in the boot process the simplefb driver is unbound and either the Pi fbdev driver is bound and gets fb0 (or the vc4 driver, depending on config).

So if you really need fb0 for your tft display, you need to block the other drivers from loading until the tft driver has loaded so it gets fb0.

tabarnakos, 2019-09-24:
So far even though the tft is properly set up as /dev/fb1, (using modprobe) I have been unsuccessful with both "startx" methods and "con2fbmap". (startx crashes, con2fbmap does nothing)

Since my device is purely accessed via ssh, I have needs for neither hdmi nor composite output, I should find a way to disable those drivers.

My next step was to look into the device tree method to see if it works better.

notro, 2019-09-24:
This should put the console on the tft: ```con2fbmap 1 1```.
As for X windows, in this file: /usr/share/X11/xorg.conf.d/99-fbturbo.conf
Change this line: ```Option          "fbdev" "/dev/fb0"```

Using Device Tree doesn't help with this problem. It doesn't affect the driver probing order. Unless you remove devices ofc.

tabarnakos, 2019-09-26:
Thanks for your help. I was able to successfully enable the display on /dev/fb1, and use it for both console of x session. I assigned it as boot console as well, and I would like to compile it into the kernel to display entire boot process now, but I will start a new thread for that.

closed 2021-08-24 by github-actions[bot]