notro/fbtft #520: Running FBTFT from staging directory from command line

mozcelikors, 2018-05-18:
I would like to activate my fbtft device automatically before kernel boot. As far as I read, this is achieved only via adding something like the following (in my case):

`fbtft_device.name=fb_ili9341 fbtft_device.gpios=reset:25,dc:24,led:18 fbtft_device.speed=16000000 fbtft_device.rotate=270 fbtft_device.bgr=1 fbtft_device.custom=1`
to
`/boot/cmdline.txt`

However, what I don't understand is there is mention of built-in branch, which I could not find, in order to accomplish this. My target uses Yocto-based Linux, which comes with fbtft from staging directory. I need to somehow change that to use built-in branch, whereever it is. I could use a couple of pointers here.

Normally, doing 

`modprobe fbtft_device custom name=fb_ili9341 gpios=reset:25,dc:24,led:18 speed=16000000 rotate=270 bgr=1`

works perfectly for me, so no issues there.

So what I've tried so far in order to move this to kernel boot are:
deactiving CONFIG_BCM2708_SPIDEV in linux-raspberrypi kernel.
and adding what I've mentioned to /boot/cmdline.txt

dmesg gives absolutely no error, in fact nothing at all related to this. Manual method (modprobe) still works fine.

Any help is greately appreciated.

notro, 2018-05-23:
> I would like to activate my fbtft device automatically before kernel boot. 

I suspect you mean before userspace comes up and udev autoloads modules and not before the kernel boots, as in activating fbtft in the bootloader.

> As far as I read, this is achieved only via adding something like the following (in my case):

That command line specifies the fbtft_device parameters, so you should be able to do: modprobe fbtft_device, and it should pick up the parameters.
It's a long time since I've tried it so I don't know it it still works.
But if it works, you can now make the fbtft_device BUILTIN instead of a loadable module and it should add the device before device probing starts.

You also need to make spi-bcm2835 and fb_ili9341 builtin as well.

mozcelikors, 2018-05-23:
I especially mean before rootfs is mounted, very early in the boot process. This is done via initial ram disk, but I thought using builtin fbtft, I can also load this very early.
The question was about builtin.. I dont know what that is? Could not find anything. Does it aimply mean that build the module into kernel? Isnt it already like that?

notro, 2018-05-23:
Builtin means that the module is linked into the kernel binary.
I don't know how you configure a Yocto kernel, but for a maunal kernel build you: ```make menuconfig``` and choose Y for builtin (marked as *), M is module.
If you use modprobe on the module it is loadable and not builtin.

closed 2021-08-24 by github-actions[bot]