notro/fbtft #523: ST7789V rotate display  0 degree 90 degree 180 degree 270 degree

DRAgon734465502, 2018-06-27:
I get this ST7789V LCD
The datasheet is here: http://tr1p1ea.net/files/downloads/hardware/lcd/SF-TC154B-8377A-N.pdf

I got the display from here: https://www.alibaba.com/product-detail/1-54-Inch-Programmable-Micro-Small_60650201984.html?spm=a2700.7724838.2017115.34.1c105d9cDJLadB

and  working with the below init code.
modprobe fbtft_device name=flexfb speed=32000000 gpios=reset:25,dc:24
modprobe flexfb setaddrwin=0 width=240 height=240 init=-1,0x11,-2,120,-1,0x36,0x00,-1,0x3A,0x05,-1,0xB2,0x0C,0x0C,0x00,0x33,0x33,-1,0xB7,0x35,-1,0xBB,0x1A,-1,0xC0,0x2C,-1,0xC2,0x01,-1,0xC3,0x0B,-1,0xC4,0x20,-1,0xC6,0x0F,-1,0xD0,0xA4,0xA1,-1,0x21,-1,0xE0,0x00,0x19,0x1E,0x0A,0x09,0x15,0x3D,0x44,0x51,0x12,0x03,0x00,0x3F,0x3F,-1,0xE1,0x00,0x18,0x1E,0x0A,0x09,0x25,0x3F,0x43,0x52,0x33,0x03,0x00,0x3F,0x3F,-1,0x29,-3

But  have a question, 
set  (0 degree) like this : 
-1,0x36,0x00,-1,
set  (90 degree) like this : 
-1,0x36,0x60,-1, 

But I can not find how to set 180 degree and 270 degree
I set 
-1,0x36,0xa0,-1,
or
-1,0x36,0xc0,-1,
just get like this picture
[image]

DRAgon734465502, 2018-06-27:
I do not why , when I set D7-MY  to 1.  the display  will lose One-third area

[image]

makermelissa, 2020-02-07:
I know this is old, but I solved it by adding the following init code to the end of `0x36,0xa0,-1,` for 180 and 270 degrees:
`0x37,0x00,0x50,-1`

So it ends up being: `0x36,0xa0,-1,0x37,0x00,0x50,-1,`.

closed 2021-08-24 by github-actions[bot]