notro/fbtft #526: ili9341﻿ and st7735s﻿ issues on NanoPi Neo Air and Armbian

troyane, 2018-10-10:
LCDs (ili9341﻿, st7735s﻿﻿) I'm able to test behave exactly the same way -- every time lighted up white screen, but nothing more.

I have NanoPi Neo Air with Armbian (5.59 Nanopiair Debian stretch 4.14.65) on it. All packages in system already updated to latest available versions. Problem remains as on stable kernel, same on nightly build.

Currently I have:
```bash
$﻿﻿ uname -a && cat /etc/issue
Linux nanopiair 4.14.74-sunxi #330 SMP Mon Oct 8 00:57:54 CEST 2018 armv7l GNU/Linux
Debian GNU/Linux 9 \n \l
```

I'm successfully loading SPI:
```bash
$ grep spi /boot/armbianEnv.txt
overlays=spi-spidev usbhost1 usbhost2﻿
param_spidev_spi_bus=0
param_spidev_spi_cs=0
```

Here is the way I'm trying to load `fbtft_device` (e.g. Waveshare 1.44 which is st7735s):
```bash
modprobe fbtft_device name=fb_st7735r speed=16000000 busnum=0 cs=0 gpios=reset:1,dc:201,led:200,cs:67 custom=1 fps=60 bgr=1 custom=1 verbose=3 width=128 height=128
```

Wiring was already rechecked for few times already.

As a result I can see next lines in `dmesg`:
```bash
[  171.665158] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[  171.670433] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[  171.671639] spidev spi0.0: spidev spi0.0 1000kHz 8 bits mode=0x00
[  171.671746] spidev spi0.0: Deleting spi0.0
[  171.672721] fbtft_device: GPIOS used by 'fb_st7735r':
[  171.672735] fbtft_device: 'reset' = GPIO1
[  171.672739] fbtft_device: 'dc' = GPIO201
[  171.672743] fbtft_device: 'led' = GPIO200
[  171.672749] fbtft_device: 'cs'﻿ = GPIO67
[  171.672765] spi spi0.0: fb_st7735r spi0.0 16000kHz 8 bits mode=0x00
[  171.684388] fb_st7735r: module is from the staging directory, the quality is unknown, you have been warned.
[  172.603112] Console: switching to colour frame buffer device 16x16
[  172.603392] graphics fb0: fb_st7735r frame buffer, 128x128, 32 KiB video memory, 4 KiB buffer memory, fps=100, spi0.0 at 16 MHz
```
And there is `/dev/fb0` available in system. But unfortunately next actions can't draw anything on screen (screen remains lighted up white without any changes):
```bash
sudo head --bytes 153600 /dev/urandom > /dev/fb0
yes | head --bytes 76800 > /dev/fb0
```

The only way to get at least some error is to run (from user of from sudo):
```bash
$ fbset -i -v
ioctl FBIOGET_VSCREENINFO: Inappropriate ioctl for device
``` 
Previously both screens were wired to RPi Zero W using almost the same scenario, and all work great.  I've got such problems only with NanoPi and Armbian.

I'm able to provide any additional information. I'll appreciate any help. Where should I dig to? Thanks.

VonHirsch, 2018-10-13:
Sorry I don't have any solution for you :(

But on a side note, would you be open to share your modprobe for ili934 on RPi Zero W?  I have a [TFT_320QVT_9341 ](http://www.icstation.com/inch-color-display-module-touch-screen-module-with-card-slot-arduino-project-p-1622.html)from icstation and I can't get it going.

I noticed there's no sample modprobe for it on the flexfb wiki page, so maybe if you have one we can convince them to add it for others using ili934.  Thx and good luck to you =)

troyane, 2018-10-14:
@VonHirsh you may refer to this article -- https://pi0cket.com/guides/ili9341-raspberry-pi-guide/
Just re-check your wirings.

VonHirsch, 2018-10-14:
@troyane TYSM!  I'll give that a shot.

VonHirsch, 2018-10-14:
@troyane actually my screen doesn't have the pins "BL, SCK, MISO or MOSI."  I've been trying to set it up in 8-bit parallel using GPIO only (no SPI) [as described here](http://hardware-libre.fr/2013/08/raspi-pip-boy-overview/).  I'm trying to use the flexpfb platform driver with fbtft (like here) but I can't get the display to do anything but turn white.  Anyway, thanks, I know this isn't a help forum =)

troyane, 2018-10-23:
@VonHirsch ili9341 should have such pins. Labels on your display module may vary, e.g. SCK = SCLK = CLK. Here is an article on how to connect ili9341 to NanoPi Neo Air -- http://trola.org/blog/ili9341-on-clean-nanopi-neo-air/

VonHirsch, 2018-10-26:
Just in case someone google's this, I got my display (TFT_320QVT_9341 from icstation) working.  I finally realized that the MCU interface mode is set to a 16-bit wide bus, which is great for speed but a pain to wire up and obviously takes up a ton of GPIO's.

After that was sorted it started up with fairly stock fbtft settings / commands & init from the wiki.  Here's what worked for me: (modify gpios per your wiring setup)

`
sudo modprobe flexfb width=320 height=240 buswidth=16 init=-1,0x28,-2,20,-1,0xCF,0x00,0x83,0x30,-1,0xED,0x64,0x03,0x12,0x81,-1,0xE8,0x85,0x01,0x79,-1,0xCB,0x39,0x2c,0x00,0x34,0x02,-1,0xF7,0x20,-1,0xEA,0x00,0x00,-1,0xC0,0x26,-1,0xC1,0x11,-1,0xC5,0x35,0x3E,-1,0xC7,0xBE,-1,0xB1,0x00,0x1B,-1,0xB6,0x0a,0x82,0x27,0x00,-1,0xB7,0x07,-1,0x3A,0x55,-1,0x36,0x38,-1,0x11,-2,120,-1,0x29,-2,20,-3
sudo modprobe --first-time fbtft_device name=flexpfb gpios=reset:25,cs:5,dc:3,wr:2,led:18,db00:6,db01:12,db02:13,db03:19,db04:16,db05:26,db06:20,db07:21,db08:14,db09:15,db10:4,db11:23,db12:24,db13:27,db14:8,db15:22
sudo fbi -d /dev/fb1 -T 1 -noverbose -a jake-halo-320-240.jpg
`

closed 2021-08-24 by github-actions[bot]