notro/fbtft #527: display works fine with manual modprobe, but not with fbtft.conf(s) at boot, but I can see boot messages with fbcon=map:10

VonHirsch, 2018-10-30:
First off thank you very much for all your work on providing fbtft and the wiki.  The wiki is really comprehensive with so many cool howtos, i've really been enjoying it.

The issue I’m having is when I load the display at boot time, I can’t get anything to display on it.  It works fine when I load manually using modprobe, but using the same exact modules & options in /etc/modules-load.d/fbtft.conf and /etc/modprobe.d/fbtft.conf the display lights up, but nothing shows when trying to display a simple image using fbi or going into xwindows.  I checked and the backlight is on, and dmesg looks clean.  No errors coming from fbi or startx.

As an additional test I added fbcon=map:10 to cmdline.txt and the console displays as it boots up, so I know the display is active.

**Drivers loaded at boot without fbcon:**
[image]

**Drivers loaded at boot with fbcon:**
[image]

**Drivers not loaded at boot time, manual modprobe commands used to load :**
[image]

Probably I’m missing something simple here (why the boot console works when drivers are loaded during boot, but not xwindows) but I can’t put my finger on it.

**/etc/modules-load.d/fbtft.conf**
```
ads7846
spi-bcm2835
spi-bcm2708
flexfb
fbtft_device
```
**/etc/modprobe.d/fbtft.conf**
```
options ads7846_device swapxy=1 cs=1 speed=2000000 model=7846
options fbtft_device name=flexpfb gpios=reset:25,cs:5,dc:3,wr:2,led:18,db00:6,db01:12,db02:13,db03:19,db04:16,db05:26,db06:20,db07:21,db08:14,db09:15,db10:4,db11:23,db1
2:24,db13:27,db14:8,db15:22
options flexfb width=320 height=240 buswidth=16 init=-1,0x28,-2,20,-1,0xCF,0x00,0x83,0x30,-1,0xED,0x64,0x03,0x12,0x81,-1,0xE8,0x85,0x01,0x79,-1,0xCB,0x39,0x2c,0x00,0x34
,0x02,-1,0xF7,0x20,-1,0xEA,0x00,0x00,-1,0xC0,0x26,-1,0xC1,0x11,-1,0xC5,0x35,0x3E,-1,0xC7,0xBE,-1,0xB1,0x00,0x1B,-1,0xB6,0x0a,0x82,0x27,0x00,-1,0xB7,0x07,-1,0x3A,0x55,-1
,0x36,0x38,-1,0x11,-2,120,-1,0x29,-2,20,-3
```

**Manual modprobe load commands**
```
sudo modprobe flexfb width=320 height=240 buswidth=16 init=\
-1,0x28,-2,20,-1,0xCF,0x00,0x83,0x30,-1,0xED,0x64,0x03,0x12,0x81,-1,0xE8,0x85,0x01,0x79,\
-1,0xCB,0x39,0x2c,0x00,0x34,0x02,-1,0xF7,0x20,-1,0xEA,0x00,0x00,-1,0xC0,0x26,-1,0xC1,0x11,\
-1,0xC5,0x35,0x3E,-1,0xC7,0xBE,-1,0xB1,0x00,0x1B,-1,0xB6,0x0a,0x82,0x27,0x00,-1,0xB7,0x07,\
-1,0x3A,0x55,-1,0x36,0x38,-1,0x11,-2,120,-1,0x29,-2,20,-3

sudo modprobe --first-time fbtft_device name=flexpfb gpios=reset:25,cs:5,dc:3,wr:2,led:18,db00:6,db01:12,db02:13,db03:19,db04:16,db05:26,db06:20,db07:21,db08:14,db09:15,db10:4,db11:23,db12:24,db13:27,db14:8,db15:22
```

**Hardware / Software**
```
TFT_320QVT_9341 & Pi Zero running Stretch
```

notro, 2018-11-03:
I don't think it can happen, but you can check to see if the framebuffer devices change index due to load order:
```
$ ls -l /sys/class/graphics/
total 0
lrwxrwxrwx 1 root root 0 Nov  3 12:06 fb0 -> ../../devices/platform/1e887000.framebuffer/graphics/fb0
lrwxrwxrwx 1 root root 0 Nov  3 12:06 fb1 -> ../../devices/platform/soc/20204000.spi/spi_master/spi0/spi0.0/graphics/fb1
lrwxrwxrwx 1 root root 0 Nov  3 12:06 fbcon -> ../../devices/virtual/graphics/fbcon
```

You can look in ```/var/log/Xorg.0.log``` to see which display device xwindows uses.

VonHirsch, 2018-11-03:
Thanks for that, framebuffer indexes check out OK (see below) and I confirmed in the log that xwindows is using fb1.

```
root@raspberrypi:/var/log# ls -l /sys/class/graphics/
total 0
lrwxrwxrwx 1 root root 0 Nov  3 12:31 fb0 -> ../../devices/platform/soc/soc:fb/graphics/fb0
lrwxrwxrwx 1 root root 0 Nov  3 12:31 fb1 -> ../../devices/platform/flexpfb.0/graphics/fb1
lrwxrwxrwx 1 root root 0 Nov  3 12:31 fbcon -> ../../devices/virtual/graphics/fbcon
```

However, I just noticed something on the fbcon display:

```failed to start load kernel modules see systemctl status```

Here's the output of systemctl status systemd-modules-load.service:

```
root@raspberrypi:/var/log# systemctl status systemd-modules-load.service

systemd-modules-load.service - Load Kernel Modules
Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static; vendor preset: enabled)
Active: failed (Result: exit-code) since Sat 2018-11-03 12:27:44 GMT; 3min 48s ago
Docs: man:systemd-modules-load.service(8)
man:modules-load.d(5)
Process: 61 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=1/FAILURE)
Main PID: 61 (code=exited, status=1/FAILURE)

Nov 03 12:27:44 raspberrypi systemd-modules-load[61]: Inserted module 'flexfb'
Nov 03 12:27:44 raspberrypi systemd-modules-load[61]: load: fbtft_device
Nov 03 12:27:44 raspberrypi systemd[1]: systemd-modules-load.service: Child 61 belongs to systemd-modules-load.service
Nov 03 12:27:44 raspberrypi systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Nov 03 12:27:44 raspberrypi systemd[1]: systemd-modules-load.service: Changed start -> failed
Nov 03 12:27:44 raspberrypi systemd[1]: systemd-modules-load.service: Job systemd-modules-load.service/start finished, result=failed
Nov 03 12:27:44 raspberrypi systemd[1]: Failed to start Load Kernel Modules.
Nov 03 12:27:44 raspberrypi systemd[1]: systemd-modules-load.service: Unit entered failed state.
Nov 03 12:27:44 raspberrypi systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
Nov 03 12:27:44 raspberrypi systemd[1]: systemd-modules-load.service: cgroup is empty
```

So i'm looking into that right now ... without much success so far but i'll keep looking.  Thx for your help.

densesimba, 2020-07-19:
would you help me with installing the drivers? i have a display with 40 pins like yours but its 2.4 inch ili9341.but i cant make it work. :( i am a totally beginner in raspberry pi.

VonHirsch, 2020-07-19:
Unless you really want to torture yourself I wouldn't bother.  You can now buy something plug & play that will work directly out of the box for roughly the same cost.  [for example this.](https://www.amazon.com/Flight-sky-Raspberry-320x480-Raspbian-RetroPie/dp/B07RQDFX3R)

densesimba, 2020-07-19:
:))))) i would torture myself because i am broke right now and i have a dead line to finish a project. I should have looked better at the product before I bought it. aaaargggghhh.

densesimba, 2020-07-19:
so pleasee mister,help a lil girl to set this up (doggy eyes)

VonHirsch, 2020-07-19:
If you have what I think you have you will want to google "ili934116 bit parallel code" - good luck, sorry that is all the assistance I can provide atm.

densesimba, 2020-07-19:
thx thats how i ve found your post :)))

closed 2021-08-24 by github-actions[bot]