notro/fbtft #530: Unknown symbol fb_deferred_io_init + fb_deferred_io_cleanup

tfindlay-au, 2019-02-06:
Hi Folks,
I have a NanoPi M4 (rk3399) with 4.4.167-rk3399 kernel.
I have been trying to build a new kernel with fbtft support, it makes the modules, installs them but when I load it via:
`sudo modprobe fbtft_device rotate=90 name=piscreen speed=16000000 gpios=reset:2,dc:71 txbuflen=32768 fps=25`
dmesg:
```
[   61.631097] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   61.631373] fbtft: Unknown symbol fb_deferred_io_init (err 0)
[   61.631391] fbtft: Unknown symbol fb_deferred_io_cleanup (err 0)
```

The screen is a FriendlyElec 2.8" with the ST7789V in it. Any tips on what I'm doing wrong ? It's been 20 years since I built a kernel from the console before, so I'm a bit rusty.

tfindlay-au, 2019-02-06:
Just for reference, modinfo show the modules as follows:

> root@nanopim4:~# modinfo fbtft
> filename:       /lib/modules/4.4.167-rk3399/kernel/drivers/staging/fbtft/fbtft.ko
> license:        GPL
> depends:
> staging:        Y
> intree:         Y
> vermagic:       4.4.167-rk3399 SMP mod_unload aarch64
> parm:           debug:override device debug level (ulong)
> parm:           dma:Use DMA buffer (bool)
> root@nanopim4:~# modinfo fb_st7789v
> filename:       /lib/modules/4.4.167-rk3399/kernel/drivers/staging/fbtft/fb_st7789v.ko
> license:        GPL
> author:         Dennis Menschel
> description:    FB driver for the ST7789V LCD Controller
> alias:          platform:st7789v
> alias:          spi:st7789v
> alias:          platform:fb_st7789v
> alias:          spi:fb_st7789v
> alias:          of:N*T*Csitronix,st7789v*
> depends:        fbtft
> staging:        Y
> intree:         Y
> vermagic:       4.4.167-rk3399 SMP mod_unload aarch64
> root@nanopim4:~#

notro, 2019-02-09:
> fbtft: Unknown symbol fb_deferred_io_init (err 0)

This indicates that ```CONFIG_FB_DEFERRED_IO``` is not enabled.
It is set by the fbtft Kconfig though: https://elixir.bootlin.com/linux/v4.4.167/source/drivers/staging/fbtft/Kconfig

closed 2019-02-28 by tfindlay-au