notro/fbtft #531: how to passing device specifc command to OLED/TFT display

rudiratlos, 2019-02-25:
I can use my OLED (SSD1306) via /dev/fb1 and console /dev/tty1
everythings works (lke drawing a line or a cirlce...)

But how can I pass device specific commands like scroll on (0x2f),scroll off (0x2e) ... ?

notro, 2019-03-03:
You can overrride the whole display controller initialization using the fbtft_device init= parameter.

Docs: https://github.com/notro/fbtft/wiki/fbtft_device#parameters
Driver default init: https://elixir.bootlin.com/linux/latest/source/drivers/staging/fbtft/fb_ssd1306.c#L31

rudiratlos, 2019-03-06:
I want to pass device specifc commands during application runtime. not at module load or application startup. I don't want to override the initialization code at all. just passing tiny commands that are really device specfic (like scroll on/off, display brightness, if there is command avail). writing directly to the spi device is not possible (blocked by the OS?)

CRImier, 2019-03-06:
same idea here. It seems to me that this is not currently possible, unless we were to implement a fbtft core "extension" of sorts, based on i.e. debugfs. I should read the code more, of course, maybe there's already such a thing - but I doubt it.

notro, 2019-03-08:
> I want to pass device specifc commands during application runtime. not at module load or application startup.

That's not implemented in fbtft.

rudiratlos, 2019-06-15:
ok, understand. Is it possible to directly access to spi device to send device specific commands. As I understand, the spi device is locked (mutex-/spinlock?) by the OS? How to bypass this?

closed 2021-08-24 by github-actions[bot]