notro/fbtft #532: no GPIO for reset

rudiratlos, 2019-02-26:
my OLED (SSD1306) works well, but I don't have the reset gpio connected. How can I disable the usage of reset pin in fbtft.conf?

cat /etc/modprobe.d/fbtft.conf
options fbtft_device name=adafruit13m debug=3 rotate=90 speed=16000000 gpios=dc:9

dmesg output:
fb_st7735r spi0.0: fbtft_request_gpios: 'dc' = GPIO9    <-- is ok
fb_st7735r spi0.0: fbtft_request_gpios: 'reset' = GPIO25 <-- I want to get RID off

notro, 2019-03-03:
This is confusing, your dmesg shows fb_st7735r not fb_ssd1306

rudiratlos, 2019-03-03:
I have 2 applications with different displays. One is a SSD1306 the other is a ST77xxx

notro, 2019-03-03:
fbtft_device can only be used for one display at a time. If you have Device Tree support, that's the best option for multiple fbtft devices.

rudiratlos, 2019-03-06:
sorry for misunderstanding. I have 2 diffrent displays. but I'm only using one at a time.
My original question is just, how to get rid of the reset gpio:
like gpios=dc:9,reset:-1
-1 is just an idea, how to omit the usage of a reset gpio pin.

CRImier, 2019-03-06:
As far as I understand, this is not possible. However, a small patch will probably fix that.

rudiratlos, 2019-03-06:
that would be great, because my displays run without any problems. no need for a reset line specially for a display. it occupies unnecessarily a gpio.

notro, 2019-03-08:
I'm suprised it doesn't work, because when you specify gpio= it should override the default gpios.
This is where gpio= is parsed in fbtft_device: https://elixir.bootlin.com/linux/latest/ident/fbtft_device_init
This is where it is applied in fbtft: https://elixir.bootlin.com/linux/latest/ident/fbtft_request_gpios

rudiratlos, 2019-03-08:
I tried two approaches to get rid of usage of reset gpio.
using reset:-1 as option will produce weired message (cut here) in dmesg, but fbtft will work.

using reset:  (omit a number) will produce an error an fbtft is not loading at all.

options fbtft_device name=sainsmart18 debug=3 rotate=90 speed=32000000 gpios=dc:9,reset:-1
fbtft will load, but have this error/warnings
[    3.432278] fb_st7735r spi0.0: 0F 1A 0F 18 2F 28 20 22 1F 1B 23 37 00 07 02 10
               0F 1B 0F 17 33 2C 29 2E 30 30 39 3F 00 07 03 10
[    3.432342] fb_st7735r spi0.0: fbtft_request_gpios: 'dc' = GPIO9
[    3.432351] ------------[ cut here ]------------
[    3.432373] WARNING: CPU: 2 PID: 149 at drivers/gpio/gpiolib.c:114 gpio_to_desc+0xc8/0xd0
[    3.432379] invalid GPIO -1

options fbtft_device name=sainsmart18 debug=3 rotate=90 speed=32000000 gpios=dc:9,reset:
fbtft is not loading at all
[    2.838597] fbtft_device: could not parse number in gpios parameter: reset:

notro, 2019-03-08:
Omitting reset should have worked: ```gpios=dc:9```. I don't understand why that isn't working for you.

rudiratlos, 2019-03-25:
ommiting the reset (just gpios=dc:9) forces fbtft to use the default reset gpio? This gpio is already used by an other function and I don't have any unused gpio left

notro, 2019-03-25:
Specifying the ```gpios=``` parameter overrides the default.

Relevant code sections from [drivers/staging/fbtft/fbtft_device.c](https://elixir.bootlin.com/linux/latest/source/drivers/staging/fbtft/fbtft_device.c):
```c
...
static char *gpios;
module_param(gpios, charp, 0000);
MODULE_PARM_DESC(gpios,
		 "List of gpios. Comma separated with the form: reset:23,dc:24 (when overriding the default, all gpios must be specified)");

...

/* Supported displays in alphabetical order */
static struct fbtft_device_display displays[] = {
...
		.name = "sainsmart18",
		.spi = &(struct spi_board_info) {
			.modalias = "fb_st7735r",
			.max_speed_hz = 32000000,
			.mode = SPI_MODE_0,
			.platform_data = &(struct fbtft_platform_data) {
				.display = {
					.buswidth = 8,
				},
				.gpios = (const struct fbtft_gpio []) {
					{ "reset", 25 },
					{ "dc", 24 },
					{},
				},
			}
		}
...

/* used if gpios parameter is present */
static struct fbtft_gpio fbtft_device_param_gpios[MAX_GPIOS + 1] = { };

...

static int __init fbtft_device_init(void)
{
...
	const struct fbtft_gpio *gpio = NULL;

...
	/* parse module parameter: gpios */
	while ((p_gpio = strsep(&gpios, ","))) {
...
		p_num = p_gpio;
		p_name = strsep(&p_num, ":");
...
		ret = kstrtol(p_num, 10, &val);
...
		strncpy(fbtft_device_param_gpios[i].name, p_name,
			FBTFT_GPIO_NAME_SIZE - 1);
		fbtft_device_param_gpios[i++].gpio = (int)val;
...
	}
	if (fbtft_device_param_gpios[0].name[0])
		gpio = fbtft_device_param_gpios;

...
	for (i = 0; i < ARRAY_SIZE(displays); i++) {
		if (strncmp(name, displays[i].name, SPI_NAME_SIZE) == 0) {
			if (displays[i].spi) {
				spi = displays[i].spi;
				spi->chip_select = cs;
				spi->bus_num = busnum;
				if (speed)
					spi->max_speed_hz = speed;
				if (mode != -1)
					spi->mode = mode;
				pdata = (void *)spi->platform_data;
...
			}

...
			if (gpio)
				pdata->gpios = gpio;
...
		}
	}

...
	if (verbose && pdata && pdata->gpios) {
		gpio = pdata->gpios;
		pr_info("GPIOS used by '%s':\n", name);
		found = false;
		while (verbose && gpio->name[0]) {
			pr_info("'%s' = GPIO%d\n", gpio->name, gpio->gpio);
			gpio++;
			found = true;
		}
		if (!found)
			pr_info("(none)\n");
	}

```

closed 2021-08-24 by github-actions[bot]