notro/fbtft #533: fbcon shortens boot output, how to avoid it?

rudiratlos, 2019-03-06:
[image]

on rpi3 with latest raspbian:
/boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=xxxxxxx-xx rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait fbcon=map:10 fbcon=font:MINI4x6 logo.nologo

/etc/modprobe.d/fbtft.conf
options fbtft_device name=adafruit13m debug=1 speed=16000000 gpios=dc:9,reset:12

every boot message ist shortend with ...
How can it be avoided? Is there a text wrap parameter somewhere ?

CRImier, 2019-03-06:
this certainly does not look like a question for this repository. Try looking it up in docs of whatever's generating this bootlog (systemd?)

rudiratlos, 2019-03-07:
I think you're right. fbcon is not the reason. I'm using tty1 as console device, that writes data to the display framebuffer.
the command:
stty -F /dev/tty1 size 
10 26
shows, that tty1 has 10 lines and 26 chars per line. It's exactly what the photo will show.
I think that the tty1 shortens the output of each line.
I try to figure out how to change tty1's parameter accordingly.

CRImier, 2019-03-07:
cool! let us know how it goes. would like to close the issue?

closed 2021-08-24 by github-actions[bot]