notro/fbtft #534: 7 inch lcd display (ssd1963 driver chip)  with windows

IFAelectro, 2019-04-15:
Hi, I have bought a 7 inch LCD display with ssd1963 driver chip. Is it possible to interface it to a windows pc (operating system would be Windows 7)?

Thanks.

troyane, 2019-04-15:
@IFAelectro It will be not possible to do via `fbtft`, since it is Linux framebuffer driver.

IFAelectro, 2019-04-15:
@troyane Thank you very much for your quick response.

IFAelectro, 2019-04-15:
@troyane Is there a way around to connect it to a Windows PC and use it as a primary display? Sorry to ask the same question again, I am a newb in this field and I was not sure whether I have expressed my problem properly.

Thanks again

IFAelectro, 2019-04-16:
I have found a solution to it. It is a different control board which has an HDMI and a VGA port.

Thanks again.

closed 2021-08-24 by github-actions[bot]