notro/fbtft #536: Adafruit pitft 3.5 (480x320)

satcom886, 2019-05-21:
Hi, I am trying to get this display to work in full resolution with the fbtft_device module, but I am having no luck. I tried these options:
1. Use fbtft_device name=adafruit28
2. Use fbtft_device name=pitft
Both of them have the same result. There is a part of the screen that I can draw onto and the rest just stays as random pixels. What should I do to get it working on full resolution? Also it may be worth mentioning, that I am on a Raspberry Pi zero w running Arch Linux ARM.

troyane, 2019-05-21:
@satcom886 take a look at https://github.com/notro/fbtft/wiki/LCD-Shields

satcom886, 2019-05-21:
So even if I do not need the touch functionality, I have to stick with the adafruit kernel (or compile my own), right?

notro, 2019-05-22:
If you're on a Raspberry Pi then you can use this DT overlay: pitft35-resistive
- https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/README#L1621-L1631
- https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/pitft35-resistive-overlay.dts

closed 2021-08-24 by github-actions[bot]