notro/fbtft #538: Cannot get ST7735 Display to work at all

RobertSmart, 2019-06-07:
Hi,

I'm banging my head against a wall here. I've been trying to get several displays to do anything on the raspberry pi. I've had them working easily on the arduino so I'm sure they work.

The display I'm trying to use is this one.

https://www.ebay.co.uk/itm/1-8-inch-ST7735-128x160-SPI-Colour-TFT-Display-Module-Arduino-Raspberry-Pi/282469570479?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160908105057%26meid%3D49362b4d9c1041c28305f65090f38962%26pid%3D100675%26rk%3D2%26rkt%3D15%26mehot%3Dag%26sd%3D302746313502%26itm%3D282469570479&_trksid=p2481888.c100675.m4236&_trkparms=pageci%3Ab6d73383-8926-11e9-8fb1-74dbd1808911%7Cparentrq%3A321705a716b0abd99f09ceb8fff2d7bb%7Ciid%3A1

I've tried using the arduino18 device driver, and I've tried flexfb as well. I don't get anything. I've tried wiring it up according to the specs under the arduino18 section and I've tried running python scripts specfic to that display and I don't get anything. 

I have never had any luck getting a display to work on raspberry pi, so starting to think I'm hexed.

I've read and reread this repo. Am I right in thinking that the drivers for fbtft are not included with the latest raspbian image? or does it still need to be installed?

If I run this

sudo modprobe fbtft_device name=adafruit18_green speed=16000000 bgr=1

I get this back.
[  653.934160] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[  653.946082] spidev spi0.0: spidev spi0.0 125000kHz 8 bits mode=0x00
[  653.946104] spidev spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
[  653.946155] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[  653.946194] spidev spi0.0: Deleting spi0.0
[  653.953478] fbtft_device: GPIOS used by 'adafruit18_green':
[  653.953495] fbtft_device: 'reset' = GPIO25
[  653.953500] fbtft_device: 'dc' = GPIO24
[  653.953504] fbtft_device: 'led' = GPIO18
[  653.953521] spidev spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
[  653.953532] spi spi0.0: fb_st7735r spi0.0 26000kHz 8 bits mode=0x00
[  654.035820] fb_st7735r: module is from the staging directory, the quality is unknown, you have been warned.
[  654.961541] graphics fb1: fb_st7735r frame buffer, 128x160, 40 KiB video memory, 4 KiB buffer memory, fps=20, spi0.0 at 26 MHz

what am I doing wrong?

julian-weinert, 2019-06-10:
I'm not sure about the speed you are using. Did you get that from the seller?
I have a different display, but to make it work, I had to use a different speed and also had to add some more parameters: `modprobe fbtft_device name=adafruit18_green gpios=reset:27,dc:25,cs:8,led:24 speed=40000000 bgr=1 fps=60 height=128 width=128 rotate=90`
Do you have your GPIOs set up the way the kernel log describes you? I have a different pinout, you can change that using the `gpios` parameter I used as well

RobertSmart, 2019-06-10:
Hi,

I've used the pinout exactly as described. I'm just trying now with level shifters to see if the display needs 5v logic to work. The display looks really dim though so I'm not holding out much hope.

RobertSmart, 2019-06-10:
Success.

I hooked the scope up to ensure it was recieving spi data, which it was, on the correct pins, so I juust messed around with the speed setting. Turns 8000000 was the magic number I needed. the colours are all inverted, but I can work with that.

Thanks,

RobertSmart, 2019-06-11:
I've got this display now working using 3.3v gpio from the raspberry pi. level shifting is not required it turns out.

However all the images are showing in negative colours. I've tried with and without the bgr flag and it doesn't seem to make any difference. are there any other settings that will effect the colours?

Thanks,

Rob

troyane, 2019-06-11:
@RobertSmart What is your full `modprobe` instrucition? Have you tried to use `adafruit18`?

RobertSmart, 2019-06-11:
Hi,

I've both of these both with and without the bgr flag set.

sudo modprobe fbtft_device name=adafruit18_green speed=8000000 bgr=1 fps=60 height=60 width=180 rotate=90 

sudo modprobe fbtft_device name=adafruit18 speed=8000000 fps=60 height=60 width=180 rotate=90

Thanks,

RobertSmart, 2019-06-26:
Had no luck getting the colours sorted, but luckily I can invert them in my code, so they display ok.

Thanks for the help.

closed 2019-06-26 by RobertSmart