notro/fbtft #539: Wrong rotation after fbcp

Creat1veM1nd, 2019-07-03:
Hello,

I am trying to build a stand alone camera with my Raspberry Pi Zero and a LCD Display. I got the display to work , it shows the desktop in the right rotation (landscape) , with the right calibration. 

I am using Kivy as programming framework to get a camera app work. I managed to start the example app. To get the app I need to run fbcp before. The app is displayed on my LCD - all well to this point. I also get the live image from the camera. The app is also in landscape mode with two buttons on the lower side of the screen. 

But: I have to touch on the right side of the screen to activate the buttons. It seems as if the app is shown in landscape mode but the screen itself seems to in portrait mode. The arrow of the touch (when I click) is shown under my pen, so I guess axis are right. 

I tried to give several rotate commands in different files (whatever I found with google I tried), but nothing helped. I also tried to find out the proper device name for my display (since I found fbtft.device=name in some posts which has to be set), with no luck. All I know is that it has "XPT2046 Touch Controller" on it and it's from Elegoo, In some files I also found "ADS7846 Touchscreen".

I am sure it's just an option somewhere I do have to set in the proper way but I seem to be unable to find the right place or the right order. I am not a pro in Linux but I do have some knowledge and of course I am willing to learn more. So I would be glad if there is a tutorial out there for this kind of problem

Is somebody able to help me ? 

Thanks in advance!

Creat1veM1nd, 2019-07-03:
I found the problem and was able to solve it. So I close that issue, since I am not able to delete it.

closed 2019-07-03 by Creat1veM1nd

CRImier, 2019-07-03:
@Creat1veM1nd can you please describe your solution?

Creat1veM1nd, 2019-07-03:
Sure, it has nothing to with fbcp, though. But with Kivy. 

I found this thread (after using totally different searchterms than before): https://www.raspberrypi.org/forums/viewtopic.php?t=145336

There is described that you have to adjust some settings in ~/.kivy/config.ini since kivy does not read the calibration file for the display. After adjusting the settings to my needs, it works as expected.