notro/fbtft #542: Is there a way to re-initialize a screen?

jamesfowkes, 2019-08-26:
I am doing development and prototyping using fbtft (using an ILI9341 based screen, though I don't think that particularly matters to the question).

Because of some issues with our test setup, it would be really useful to be able to re-initialise the attached screen if it gets power cycled. Currently doesn't work being power cycled as it won't be re-initialised by fbtft, which (understandably) assumes everything is fine.

Is there a way to do this?

notro, 2019-08-26:
You can either unload and reload the driver module, or you can unbind and (re)bind the driver to the device:
```
$ ls -l /sys/bus/spi/drivers/<driver>/*bind
--w------- 1 root root 4096 Aug 26 15:06 /sys/bus/spi/drivers/<driver>/bind
--w------- 1 root root 4096 Aug 26 15:06 /sys/bus/spi/drivers/<driver>/unbind
```
If you can use any of the ili9341 drm drivers, they will reinitialize if needed when the display pipeline is enabled:
https://elixir.bootlin.com/linux/v5.2/source/drivers/gpu/drm/tinydrm/ili9341.c
https://elixir.bootlin.com/linux/v5.2/source/drivers/gpu/drm/tinydrm/mi0283qt.c

jamesfowkes, 2019-08-26:
OK, so just to check, the commands for bind and unbind would be (for my screen on SPI0.0):

```
echo "spi0.0" | /sys/bus/spi/drivers/fb_ili9341/bind
echo "spi0.0" | /sys/bus/spi/drivers/fb_ili9341/unbind
```

Is that correct?

notro, 2019-08-26:
Yeah, I think so.

closed 2021-08-24 by github-actions[bot]