notro/fbtft #543: module param gpios not recognized in mainline 5.3.0, broken since 5.1.rc1

yam001, 2019-09-17:
The error was: fbtft_device: unknown parameter 'gpios' ignored
Looking at the code, somebody removed the gpios module_param definition from fbtft_device.c since 5.1.rc1 and it has been broken ever since. Missing code:
static char *gpios;
module_param(gpios,

Do you know what the new interface should be? I have been using the commands like the following:
sudo modprobe fbtft_device custom name=fb_ili9341 gpios=reset:1,dc:201,led:6 speed=16000000 txbuflen=32768 rotate=270 bgr=1

The parameter gpios is not being recognized any more. What should it be replaced with to specify reset, dc, and led? Thanks.

notro, 2019-09-17:
I have looked at it and fbtft now only supports Device Tree for adding devices. fbtft_device is now useless since all the gpio info is gone. 
fbtft_device should never have been added to staging in the first place. Device Tree, ACPI and platform code is where such info should reside. The kernel doesn't have modules like fbtft_device.
If the staging maintainer knew what fbtft_device was doing, it probably wouldn't have been been merged in the first place. 
So since it's now broken I guess it should be removed to avoid confusion.

yam001, 2019-09-17:
Thanks for the quick reply, I really do appreciate your work and your tireless support effort here. One more question, do you know if in the device tree, the reference to *-gpios would still work - as an example in the following device tree fragment (reset-gpios, dc-gpios, led-gpios):

                                spi-max-frequency = <32000000>;
				txbuflen = <32768>;
				rotate = <90>;
				bgr = <1>;
				fps = <30>;
				buswidth = <8>;
				reset-gpios = <&pio 0 1 0>; /*PA1*/
				dc-gpios = <&pio 6 9 0>; /*PG9*/
				led-gpios = <&pio 0 6 1>; /*PA6*/
				debug = <0>;

I have not got it quite working yet but its turning on the led. Thanks again.

notro, 2019-09-17:
Yeah that should work even though I haven't tried fbtft in years myself. You use 32MHz here but 16MHz in your previous example. Increasing the speed could cause problems, I don't know if this applies in your case.

satmandu, 2020-02-13:
I'd add here that with a 5.3 kernel with the 5.4 fbtft module (and an updated display dtbo file) I get a working display and touchscreen, but for kernel 5.4 and onwards I get a working display but non-working touchscreen.

closed 2021-08-24 by github-actions[bot]