notro/fbtft #544: Generic ili9341 display Kernel compilation on RPi Zero via SPI1

tabarnakos, 2019-09-26:
This is a follow-up of [this thread](https://github.com/notro/fbtft/issues/518
)
So I've successfully integrated my 2.2" display using the "modprobe" method, but I'm not quite satisfied yet with how late in the boot process the display turns on, approx 50sec on the Pi Zero.

Here is the command that I am using to mount the display:
in /etc/modprobe.d/fbtft.conf:
`options fbtft_device name=rpi-display speed=10000000 gpios=reset:5,dc:6,led:13 rotate=90 cs=0 busnum=1`
	
in /etc/modules-load.d/fbtft.conf:
`spi-bcm2835aux`
`fbtft_device`

So to make the driver load earlier in the boot process, I understand that I should compile it directly into the kernel. I've successfully downloaded, cross-compiled and updated the kernel with success, but I can't seem to figure out which changes need to be made to enable:

1.  The device overlay for this specific display: 320x240, ili9341 chip,  no touchscreen, SPI1 port, CE0
2.  How to load it into the PI Zero. [These](https://github.com/notro/fbtft/wiki/Build-kernel-and-fbtft-drivers#configure-devices) instructions seem to be for different SoC bcm2708, whereas the PI zero has a bcm2835. The bcm_register_device function is not found in the board_bcm2835.c file includes, and I am not quite sure how to port it to the Pi Zero.

Any insights or guide on how to integrate this display into the kernel is highly welcome.

notro, 2019-09-29:
Those instructions are out of date (I've deleted them now). 
You need to make a Device Tree overlay so that the fbtft spi devices are available at boot. Some info here: https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

You can adapt this overlay: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/rpi-display-overlay.dts
Remove the overrides, spidev and touchcontroller parts. You can also remove the pins node since you don't need to set pullup on input pins.

Maybe something like this:
```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835";

	fragment@1 {
		target = <&spi1>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			generic22-display@0{
				compatible = "ilitek,ili9341";
				reg = <0>;

				spi-max-frequency = <10000000>;
				rotate = <90>;
				//bgr; not sure if you need this
				reset-gpios = <&gpio 5 0>;
				dc-gpios = <&gpio 6 0>;
				led-gpios = <&gpio 13 1>;
				debug = <0>;
			};
		};
	};
};
```
I assume you enable spi1 using its DT overlay.

tabarnakos, 2019-10-17:
Thank you ! I needed to add "buswidth = <8>;" to the device tree, but otherwise worked like a charm.

However the driver is now loaded even later in the boot process, only when the X session starts.

notro, 2019-10-20:
When you're adding devices using a DT overlay and load/specify the overlay in /boot/config.txt, then the devices are present when the kernel boots. This means that the driver load order decides when the display lights up. You depend on 2 drivers: spi-bcm2835aux and fb_ili9341. Both are compiled as modules on stock Raspbian. This means that they will load when userspace is up (udev/systemd). If you want it to load earlier, you have to either build your own kernel and make the drivers builtin (built into the kernel binary) or put the driver modules in initramfs (out-of-date info: https://github.com/notro/fbtft/wiki/Bootsplash#setup-initramfs).

closed 2021-08-24 by github-actions[bot]