notro/fbtft #545: JDI MIP display (LPM027M128C) | help with modprobe fbtft_device syntax

mrzachsnyder, 2019-09-28:
Hello notro, thank you very much for FBTFT and being willing to help people use it. I’ve got a fancy 2.7” JDI MIP display ([LPM027M128C](https://www.j-display.com/english/product/reflective.html)) that I’m trying to get working with a Pi Zero W. It’s hooked up correctly enough that I can get a python script to display a color bar on it, but figuring out how to actually use it as a monitor has been quite difficult.

[image]
[reference: how I figured out the color bar](https://os.mbed.com/questions/85441/Is-LPM013M126A-worked-by-python-in-raspb/)

The pinout on this display is kinda weird (or I'm just a noob), not really consistent with more hobbyist-friendly displays that you’d usually use with the Pi. Here's how I've connected things.

[image]

I've used several variations of this command, getting the same result every time. This is the most minimal version of the command that gives me the result described below.

`sudo modprobe fbtft_device name=itdb28_spi speed=2000000`

I'm embarrassed to say that I've tried just about every gpio= option that I can think of, but it doesn't seem to affect anything. I chose itdb28_spi because it seems to be the most similar to my display? I've probably read your [fbtft_device wiki](https://github.com/notro/fbtft/wiki/fbtft_device) ten times now but I can't figure out which pins on my display correspond to the dc,reset,cs,etc pins that you describe in the wiki. The only thing I've figured out is that speed matters - setting to 8MHz doesn't work. It's supposed to be 2MHz but 2.5MHz will still give me the flickering. Anyways yeah my lack of experience with how displays work is showing... I think I'm close but I'm really struggling with figuring out which troubleshooting lever to pull from here.

Info that might be helpful to you...
```
ls -l /dev/fb*
crw-rw---- 1 root video 29, 0 Sep 27 17:02 /dev/fb0
crw-rw---- 1 root video 29, 1 Sep 27 17:03 /dev/fb1
```

```
fbset -fb /dev/fb1
mode "240x320"
    geometry 240 320 240 320 16
    timings 0 0 0 0 0 0 0
    nonstd 1
    rgba 5/11,6/5,5/0,0/0
endmode
```

`cat /dev/urandom > /dev/fb1`
[video of how the screen responds
](https://www.youtube.com/watch?v=B_1gwSQOMsg&feature=youtu.be)

`sudo FRAMEBUFFER=/dev/fb1 startx`
[video of how the screen responds](https://www.youtube.com/watch?v=Jyo4jFD_JsU&feature=youtu.be)

Very much looking forward to hearing suggestions and getting this thing working! From what I can tell this would be the first JDI MIP display to work with your drivers. Thanks again for the help.

notro, 2019-09-29:
I'm sorry but there are no fbtft drivers that support this display. Looking at the datasheet it has a 2 byte prefix on transfers. 6 bits for mode and optionally 10 bits for line address (or dummy).

You would have to write a custom driver that overrides fbtft defaults, something along the lines of this one: https://elixir.bootlin.com/linux/latest/source/drivers/staging/fbtft/fb_watterott.c

If you're able to drive the display from userspace, you could mmap the fbdev memory and just read it at fixed intervals and write the contents to your display.
Or you could use https://github.com/tasanakorn/rpi-fbcp to get the framebuffer contents.

mrzachsnyder, 2019-09-30:
Thanks for the reply, I suspected as much but it's great to have that confirmed. If I had more time I'd dive into the weeds and try to write a custom driver - maybe I'll come back to this someday because the display is pretty neat. For now I'll just keep using python.

For the record, here's a quick and dirty modification of the original MIP_ColorBar.py script. It just cycles through three different ways of creating the color bars, which ended up being enough for us to decide to go with a different display option.

```
import spidev
import time
import RPi.GPIO as GPIO
import random

channel = 0
disp = 13
scs = 15
VcomSel = 11

spi = spidev.SpiDev()
spi.open(0, 1)
spi.mode = 0b00     #SPI MODE0
spi.max_speed_hz=2000000   #MAX 2MHz
spi.no_cs
time.sleep(0.1)     #Wait

GPIO.setmode(GPIO.BOARD)
GPIO.setup(disp   , GPIO.OUT)
GPIO.setup(scs    , GPIO.OUT)
GPIO.setup(VcomSel, GPIO.OUT)

GPIO.output(scs, 0)     #1st=L
GPIO.output(disp, 1)    #1st=Display On
#GPIO.output(disp, 0)   #1st=No Display
GPIO.output(VcomSel, 0) #L=VCOM(1Hz)
time.sleep(0.1)

GPIO.output(scs, 1)
time.sleep(0.000006)
spi.xfer2([0x20,0]) # ALL CLEAR MODE
GPIO.output(scs, 0)
time.sleep(0.000006)

print("------------------")
print("USEFUL INFO: START")

bars = 16
asdf = 0
foo = 4
for x in range(36):
        if asdf >= 192:
                asdf = bars
                foo = foo-1
        else:
                asdf = asdf + bars
        #def colorbar_generation():
        #Color Bar
        cmd_buff = bytearray(202)
        cmd_buff[0] = 0x90  #'1x01xxyy=4bit mode,
        cmd_buff[1] = 0x00  #Line
        #print(cmd_buff[1])  #[0] prints 144, [1] prints 0
        for i in range(asdf):
                #print(asdf)
                k=i>>foo
                cmd_buff[i+2] = (k<<5)&0xe0 | (k<<1)&0xe #R0,G0,B0,D0,R1,G1,B1,D1
                #cmd_buff[i+2] = (k<<1)&0xe | (k<<5)&0xe0 #R0,G0,B0,D0,R1,G1,B1,D1

        #Buffer->MIP
        GPIO.output(scs, 1)
        time.sleep(0.000006)
        for j in range(240):
                cmd_buff[1] = j  #line(M128A=240lines)
                resp = spi.xfer2(cmd_buff)
        time.sleep(0.000006)
        GPIO.output(scs, 0)
        #commenting
        #time.sleep(0.1)

print("USEFUL INFO: STOP")
print("-----------------")

#BackLight ----------------------------------------　May 30, 2019
GPIO.setup(7,GPIO.OUT)      #Pin7 = GPIO4
p7=GPIO.PWM(7,60)
p7.start(0)
for x in range(100):
    for pw in range(0,100,1):
        p7.ChangeDutyCycle(pw)
        time.sleep(0.05)
    for pw in range(100,0,-1):
        p7.ChangeDutyCycle(pw)
        time.sleep(0.05)

p7.stop()
#-----------------------------------------------------

spi.close()

GPIO.output(disp, 1)
#time.sleep(0.1)

#GPIO.cleanup()
```

closed 2019-09-30 by mrzachsnyder