notro/fbtft #546: waveshare35a works with dtoverlay on kernel 5.0, but not on kernel 5.3 on a rpi 3b+?

satmandu, 2019-10-02:
It appears the dt overlays are being loaded, but the screen doesn't do anything when I boot on 5.3.

There's a ticket open https://github.com/raspberrypi/linux/issues/3263 where it is suggested that the overlays are being loaded correctly.

Is there an issue with 5.3 and using displays with fbtft?

notro, 2019-10-05:
There was changes to the fbtft gpiop handling in 5.3-rc1 that was only compile tested so it broke stuff. I see there's been some fixes in 5.3-rc4 to rectify this.
Problem commit: [Staging: fbtft: Switch to the gpio descriptor interface](https://github.com/torvalds/linux/commit/c440eee1a7a1d0f2d5fc2ee6049e4a05da540f01)
History: https://github.com/torvalds/linux/commits/master/drivers/staging/fbtft

Maybe there's still fallout, I don't know. Currently fbtft is unmaintained, it's listed as ```Orphan``` in [MAINTAINERS](https://github.com/torvalds/linux/blob/v5.3/MAINTAINERS#L6210)

satmandu, 2019-10-30:
I tried building 5.4's fbtft as a module for 5.3.x, and have the dtoverlay being loaded with 
https://github.com/swkim01/waveshare-dtoverlays/blob/master/waveshare35a.dts modified so thaat 
```
                                reset-gpios = <&gpio 25 0>;
                                dc-gpios = <&gpio 24 0>;
```
is now 

```
                                reset-gpios = <&gpio 25 1>;
                                dc-gpios = <&gpio 24 1>;
```
(since apparently that may get flipped in newer fbtft.)

That dtb still loads fine in 4.19 but doesn't do anything in anything newer.

satmandu, 2019-10-31:
The winning combo appears to be modifying https://github.com/swkim01/waveshare-dtoverlays/blob/master/waveshare35a.dts thus:
```
--- waveshare35a.dts    2019-09-30 09:17:06.000000000 -0400
+++ waveshare35a.dts 2019-10-30 15:23:48.000000000 -0400
@@ -54,7 +54,7 @@
                                fps = <30>;
                                buswidth = <8>;
                                regwidth = <16>;
-                               reset-gpios = <&gpio 25 0>;
+                               reset-gpios = <&gpio 25 1>;
                                dc-gpios = <&gpio 24 0>;
                                debug = <0>;
 
@@ -81,7 +81,7 @@
                                spi-max-frequency = <2000000>;
                                interrupts = <17 2>; /* high-to-low edge triggered */
                                interrupt-parent = <&gpio>;
-                               pendown-gpio = <&gpio 17 0>;
+                               pendown-gpio = <&gpio 17 1>;
                                ti,x-plate-ohms = /bits/ 16 <60>;
                                ti,pressure-max = /bits/ 16 <255>;
```

Additionally I had to make sure that for the kernel config
this was set to module: ```CONFIG_SPI_BCM2835AUX=m```

and then also compile the linux 5.4 series fbtft modules.

closed 2019-10-31 by satmandu