notro/fbtft #548: unload fbtft, reload spidev, spidev0.0 does not reappear

mmueller-kaffeeschluerfercom, 2019-11-11:
I have posted my issue here: [https://stackoverflow.com/questions/58730774/raspbian-spi-tft-display-how-to-disable-fbtft-at-runtime-and-restore-reenable](https://stackoverflow.com/questions/58730774/raspbian-spi-tft-display-how-to-disable-fbtft-at-runtime-and-restore-reenable) so I hope it's ok and not offending anyone if I ask the question here again.

There's someone else with the same issue here: [https://www.raspberrypi.org/forums/viewtopic.php?t=178252](https://www.raspberrypi.org/forums/viewtopic.php?t=178252).

It may also be related to the closed issue #101...

I want to show a splash screen as soon as possible when booting (I believe the best way to achieve this is to load fbtft and to copy a raw RGB565 splash screen file to /dev/fb1, hope I'm right). But then, after the boot process, my own program talks directly to /dev/spidev0.0, so I need to restore the device.

In the wiki on fbtft_device, it says: 

> Then it deletes the device connected to spi0.0 (spidev) so we can register a new one.

Is the device restored when I do a modprobe -r fbtft_device ?

I tried unloading and reloading everything:
```
modprobe -r flexfb
modprobe -r fbtft_device
modprobe -r spidev
modprobe -r spi_bcm2835
modprobe spidev
modprobe spi_bcm2835
```
without success. The device /dev/spidev0.1 is there, but not /dev/spidev0.0. Until now, I have to reboot to restore /dev/spidev0.0. Is it possible to restore it at runtime?

CRImier, 2019-11-11:
Hi! I'm not the author of the repo, but I've been thinking about this exact scenario, as I'm facing the same problem. Haven't yet had time to try it but [can you try this?](https://github.com/notro/fbtft/issues/542#issuecomment-524911845) I'll try it too when I have the time.

notro, 2019-11-11:
fbtft_device doesn't restore the spidev device.
Can you put your program in initramfs and use that to display a bootspslash, avoiding fbtft altogether?

mmueller-kaffeeschluerfercom, 2019-11-11:
Well, I never thought about this because /dev/spidev0.0 is populated so late in the startup process that I thought it is dependent on a lot of things. I wasn't aware that I could load modules like spidev and start my own programs in initrd...

But yes, I can make a program displaying the splash image directly with spidev. That would mean loading spi_bcm2835 and spidev in initramfs and add my program in a startup script? Will try that! Thanks!

Out of curiosity: 
> fbtft_device doesn't restore the spidev device.

Is there still a way to restore it without rebooting?

@CRImier I tried to bind the device with `echo "spi0.0" | /sys/bus/spi/drivers/spidev/bind` as root but I get "bash: /sys/bus/spi/drivers/spidev/bind: Permission denied".

mmueller-kaffeeschluerfercom, 2019-11-11:
@CRImier of course it should be `echo "spi0.0" > /sys/bus/spi/drivers/spidev/bind` but then I get `bash: echo: write error: No such device`

notro, 2019-11-11:
> Is there still a way to restore it without rebooting?

You could create a Device Tree overlay that adds a spidev device and load it at runtime using the ```dtoverlay``` command.

mmueller-kaffeeschluerfercom, 2019-11-13:
Thank you very much. I managed to have a boot logo with my own program in an initramfs. Thanks for the idea!

closed 2019-11-13 by mmueller-kaffeeschluerfercom