notro/fbtft #551: 3.5" (480x320) ILI9486 SPI Touch Panel

Int32Overflow, 2020-01-28:
Hello,

I have purchased the following TFT panel from Aliexpress: `3.5" (480x320) ILI9486 SPI Touch Panel` (https://www.aliexpress.com/item/33005785723.html)

I can't get it to work :-( 
```
<file /etc/modprobe.d/fbtft.conf >
options fbtft_device custom name=fb_ili9486 speed=1000000 gpios=reset:25,dc:24,led:18 width=480 height=320 buswidth=8 fps=10 verbose=3
```

```
<file:  /var/log/messages >
Jan 28 12:32:58 raspberrypi kernel: [    4.491478] spi spi0.0: spidev spi0.0 125000kHz 8 bits mode=0x00
Jan 28 12:32:58 raspberrypi kernel: [    4.497716] spi spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
Jan 28 12:32:58 raspberrypi kernel: [    4.497787] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
Jan 28 12:32:58 raspberrypi kernel: [    4.509310] spi spi0.0: Deleting spi0.0
Jan 28 12:32:58 raspberrypi kernel: [    4.510545] fbtft_device: GPIOS used by 'fb_ili9486':
Jan 28 12:32:58 raspberrypi kernel: [    4.518470] fbtft_device: 'reset' = GPIO25
Jan 28 12:32:58 raspberrypi kernel: [    4.522620] fbtft_device: 'dc' = GPIO24
Jan 28 12:32:58 raspberrypi kernel: [    4.526501] fbtft_device: 'led' = GPIO18
Jan 28 12:32:58 raspberrypi kernel: [    4.530525] spi spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
Jan 28 12:32:58 raspberrypi kernel: [    4.530542] spi spi0.0: fb_ili9486 spi0.0 1000kHz 8 bits mode=0x00
Jan 28 12:32:58 raspberrypi kernel: [    5.224121] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Jan 28 12:32:58 raspberrypi kernel: [    6.113137] fb_ili9486: module is from the staging directory, the quality is unknown, you have been warned.
Jan 28 12:32:58 raspberrypi kernel: [    8.985538] Console: switching to colour frame buffer device 60x40
Jan 28 12:32:58 raspberrypi kernel: [    8.987743] graphics fb1: fb_ili9486 frame buffer, 480x320, 300 KiB video memory, 4 KiB buffer memory, fps=10, spi0.0 at 1 MHz
```

Can anyone help me?

mahathug, 2020-05-03:
Can you verify that it's a ili9486? I ordered the same and got ili9488 instead. Also few of the comments says that even they got ili9488. 
If you have already got it working, please share the procedure.

mahathug, 2020-05-17:
If it's a ili9488 
https://github.com/mahathug/linux_4.20/blob/master/fb_ili9488.patch

if it's a ili9486, tinydrm version of it is available in mainline now.

closed 2021-08-24 by github-actions[bot]