notro/fbtft #552: Lichee Pi Nano ILI9341 LCD White

piyushverma, 2020-03-30:
Hi Notro,

Thanks for your great job of FBTFT.

I am trying to use ILI9341 LCD with FBTFT but I am getting white lcd which looks initialization not success.

I have verified LCD & Connection with STM32 it's working out of box with uglib.

In Lichee Pi, I have check with oscilloscope that All pin valid signal comping out like SPI_CLK, SPI_MOSI, CD, RESET.

I have modified device tree as below.

+&spi0 {
+       status = "okay";
+
+       ili9341@0 {
+               compatible = "ilitek,ili9341";
+               reg = <0>;
+
+               spi-max-frequency = <15000000>;
+               rotate = <270>;
+               bgr;
+               fps = <10>;
+               buswidth = <8>;
+               reset-gpios = <&pio 1 7 GPIO_ACTIVE_LOW>;
+               dc-gpios = <&pio 1 5 GPIO_ACTIVE_LOW>;
+               debug = <0>;
+       };
+};

/dev/fb0 node is created.

Any suggestion how to debug this ?

Thanks

notro, 2020-04-03:
I would set ```debug = <7>;``` and look in the kernel log to verify that the initialization is the same as that library you mention.

piyushverma, 2020-04-11:
Hi Notro,
Thanks for your reply, & sorry I missed your message.
I don't have experiance working with spi display,
here what I have found ucglib init & fbtft log, can you check them and suggest how to fix ?

Here is code what I found in ucglib
  UCG_CFG_CD(0,1),				/* DC=0 for command mode, DC=1 for data and args */
  UCG_RST(1),					
  UCG_CS(1),					/* disable chip */
  UCG_DLY_MS(5),
  UCG_RST(0),					
  UCG_DLY_MS(5),
  UCG_RST(1),
  UCG_DLY_MS(50),
  UCG_CS(0),					/* enable chip */
  
  UCG_C10(0x011),				/* sleep out */
  UCG_DLY_MS(10),
  //UCG_C10(0x038),				/* idle mode off */
  UCG_C10(0x013),				/* normal display on */

  //UCG_C14(0x0ed, 0x055, 0x001, 0x023, 0x001), 	/* power on sequence control (POR values) */
  //UCG_C11(0x0f7, 0x020), 		/* pump ratio control (POR value) */
  

  UCG_C10(0x20), 				/* not inverted */
  //UCG_C10(0x21), 				/* inverted */

  UCG_C11(0x03a, 0x066), 		/* set pixel format to 18 bit */
  //UCG_C11(0x03a, 0x055), 		/* set pixel format to 16 bit */

  //UCG_C12(0x0b1, 0x000, 0x01b), 	/* frame rate control (POR values) */

  UCG_C14(0x0b6, 0x00a, 0x082 | (1<<5), 0x027, 0x000), 	/* display function control (POR values, except for shift direction bit) */
  
  //UCG_C11(0x0b7, 0x006), 		/* entry mode, bit 0: Low voltage detection control (0=off) */

  UCG_C11(0x0c0, 0x021), 		/* power control 1 (reference voltage level), POR=21 */
  UCG_C11(0x0c1, 0x002), 		/* power control 2 (step up factor), POR=2 */
  UCG_C11(0x0c7, 0x0c0), 		/* VCOM control 2, enable VCOM control 1 */
  UCG_C12(0x0c5, 0x031, 0x03c), 	/* VCOM control 1, POR=31,3C */
  
  UCG_C15(0x0cb, 0x039, 0x02c, 0x000, 0x034, 0x002), 	/* power control A (POR values) */
  UCG_C13(0x0cf, 0x000, 0x081, 0x030), 	/* power control B (POR values) */

  UCG_C13(0x0e8, 0x084, 0x011, 0x07a), 	/* timer driving control A (POR values) */

  UCG_C12(0x0ea, 0x066, 0x000), 	/* timer driving control B (POR values) */
  //UCG_C12(0x0ea, 0x000, 0x000), 		/* timer driving control B  */

  //UCG_C10(0x28), 				/* display off */
  //UCG_C11(0x0bf, 0x003), 		/* backlight control 8 */
  UCG_C10(0x029), 				/* display on */
  //UCG_C11(0x051, 0x07f), 		/* brightness */
  //UCG_C11(0x053, 0x02c), 		/* control brightness */
  //UCG_C10(0x028), 				/* display off */

  UCG_C11( 0x036, 0x008),
  
  UCG_C14(  0x02a, 0x000, 0x000, 0x000, 0x0ef),              /* Horizontal GRAM Address Set */
  UCG_C14(  0x02b, 0x000, 0x000, 0x001, 0x03f),              /* Vertical GRAM Address Set */
  UCG_C10(  0x02c),               /* Write Data to GRAM */

  
  UCG_CS(1),					/* disable chip */
  UCG_END(),					/* end of sequence */

Here is fbtft initialization

[    0.934104] fbtft_of_value: buswidth = 8
[... 54 lines cut ...]
[    1.380655] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 26
[    1.386264] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 01
[    1.391935] fb_ili9341 spi1.0: fbtft_write_reg8_bus8: b7 07
[    1.397636] fb_ili9341 spi1.0: fbtft_write_spi(len=1): b7
[    1.403239] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 07
[    1.408894] fb_ili9341 spi1.0: fbtft_write_reg8_bus8: b6 0a 82 27 00
[    1.415298] fb_ili9341 spi1.0: fbtft_write_spi(len=1): b6
[    1.420949] fb_ili9341 spi1.0: fbtft_write_spi(len=4): 0a 82 27 00
[    1.427392] fb_ili9341 spi1.0: fbtft_write_reg8_bus8: 11
[    1.432765] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 11
[    1.538260] fb_ili9341 spi1.0: fbtft_write_reg8_bus8: 29
[    1.543634] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 29
[    1.569296] fb_ili9341 spi1.0: fbtft_write_reg8_bus8: 36 48
[    1.574931] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 36
[    1.580583] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 48
[    1.586159] fb_ili9341 spi1.0: fbtft_update_display(start_line=0, end_line=319)
[    1.593592] fb_ili9341 spi1.0: fbtft_write_reg8_bus8: 2a 00 00 00 ef
[    1.600045] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 2a
[    1.605631] fb_ili9341 spi1.0: fbtft_write_spi(len=4): 00 00 00 ef
[    1.612083] fb_ili9341 spi1.0: fbtft_write_reg8_bus8: 2b 00 00 01 3f
[    1.618567] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 2b
[    1.624177] fb_ili9341 spi1.0: fbtft_write_spi(len=4): 00 00 01 3f
[    1.630623] fb_ili9341 spi1.0: fbtft_write_reg8_bus8: 2c
[    1.635992] fb_ili9341 spi1.0: fbtft_write_spi(len=1): 2c
[    1.641628] fb_ili9341 spi1.0: fbtft_write_vmem16_bus8(offset=0, len=153600)

notro, 2020-04-12:
Here's the initialization sequence side by side:

```
This is soft reset
fbtft_write_reg8_bus8: 01

//UCG_C10(0x28), /* display off /
fbtft_write_reg8_bus8: 28

UCG_C13(0x0cf, 0x000, 0x081, 0x030), / power control B (POR values) */
fbtft_write_reg8_bus8: cf 00 83 30

//UCG_C14(0x0ed, 0x055, 0x001, 0x023, 0x001), /* power on sequence control (POR values) /
fbtft_write_reg8_bus8: ed 64 03 12 81

UCG_C13(0x0e8, 0x084, 0x011, 0x07a), /* timer driving control A (POR values) */
fbtft_write_reg8_bus8: e8 85 01 79

UCG_C15(0x0cb, 0x039, 0x02c, 0x000, 0x034, 0x002), /* power control A (POR values) /
fbtft_write_reg8_bus8: cb 39 2c 00 34 02

//UCG_C11(0x0f7, 0x020), / pump ratio control (POR value) */
fbtft_write_reg8_bus8: f7 20

UCG_C12(0x0ea, 0x066, 0x000), /* timer driving control B (POR values) /
//UCG_C12(0x0ea, 0x000, 0x000), / timer driving control B */
fbtft_write_reg8_bus8: ea 00 00

UCG_C11(0x0c0, 0x021), /* power control 1 (reference voltage level), POR=21 /
fbtft_write_reg8_bus8: c0 26

UCG_C11(0x0c1, 0x002), / power control 2 (step up factor), POR=2 /
fbtft_write_reg8_bus8: c1 11

UCG_C12(0x0c5, 0x031, 0x03c), / VCOM control 1, POR=31,3C */
fbtft_write_reg8_bus8: c5 35 3e

UCG_C11(0x0c7, 0x0c0), / VCOM control 2, enable VCOM control 1 /
fbtft_write_reg8_bus8: c7 be

UCG_C11(0x03a, 0x066), /* set pixel format to 18 bit /
//UCG_C11(0x03a, 0x055), / set pixel format to 16 bit */
fbtft_write_reg8_bus8: 3a 55

//UCG_C12(0x0b1, 0x000, 0x01b), /* frame rate control (POR values) */
fbtft_write_reg8_bus8: b1 00 1b

fbtft_write_reg8_bus8: 26 01

//UCG_C11(0x0b7, 0x006), /* entry mode, bit 0: Low voltage detection control (0=off) */
fbtft_write_reg8_bus8: b7 07

UCG_C14(0x0b6, 0x00a, 0x082 | (1<<5), 0x027, 0x000), /* display function control (POR values, except for shift direction bit) */
fbtft_write_reg8_bus8: b6 0a 82 27 00

UCG_C10(0x011), /* sleep out /
fbtft_write_reg8_bus8: 11

UCG_C10(0x029), / display on /
fbtft_write_reg8_bus8: 29

UCG_C11( 0x036, 0x008),
fbtft_write_reg8_bus8: 36 48

```

You can override the initialization using the ```init=``` Device Tree property.

Example: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/mz61581-overlay.dts

The command are OR'ed with 0x1000000, delay are OR'ed with 0x2000000, parameters are as-is.

piyushverma, 2020-06-01:
Finally fix the issue, 
I put LCD CS to pull low in hardware.
By using logic analyzer I notice linux is spitting some strange data in spi but CS goes high on that time. so SPI CS was not in sync with linux, I connected additional line from linux CS pin to lcd and then all work out of box.

Hope this information can help other.

closed 2020-06-01 by piyushverma

mzashh, 2023-07-13:
@piyushverma i am trying to do something similar but i am also getting a white screen i noticed you used 
reset-gpios = <&pio 1 7 GPIO_ACTIVE_LOW>;
           dc-gpios = <&pio 1 5 GPIO_ACTIVE_LOW>;