notro/fbtft #554: Wiki is lacking since fbtft_device has been removed

muesli4, 2020-05-10:
Since `fbtft_device` was removed I can no longer use my display. I tried to find a solution that involves device tree overlays but the documentation is so high level that I couldn't find anything useful in 2 hours. Before I could easily connect a display with custom pins. What is the proposed solution now?

Sindbad6, 2020-07-20:
Yeah, same to me. Would be nice to explain in the Wiki, how to get it working in the device tree.
Thanks alot in advanced.

Broodie, 2020-11-13:
Maybe you can try this out: https://github.com/Broodie/ili9341_tft_ubuntu_20_04
I've just managed to make an ili9341 display working.

muesli4, 2021-08-16:
The issue is still present.

notro, 2021-11-07:
I've updated the wiki and added an overlay that partially replaces fbtft_device: https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

muesli4, 2021-11-17:
@notro, I appreciate your effort.  However, in terms of usability I think this is devastating compared to the old module.  I do not even understand what half of that stuff does.  Now everyone that deviates a slight bit by not purchasing ready-made modules is forced to deal with this crap.  All I want to do is assign pins and display settings.  Where is the abstraction for details here?  This is frustrating and deters people from doing great stuff.

How would I convert from module arguments to a dts file?  I'm afraid a partial replacement is not sufficient.

muesli4, 2021-12-08:
I was forced to do a system update and consequently had to read `/boot/overlays/README` again where I found the `fbtft` overlay.  This is specified in the wiki and if I would have paid more attention to the wall of text then I might have found it sooner.  In that sense, I would like to apologize for my last comment that I posted out of sheer frustration.

In the end the effort was minuscule and it was easy to simply adapt the parameters in the same way the kernel module received them.

If I may suggest some improvements for readability:
* Separate the wall of text into different sections. _I seem to not to be the only person that managed to simply read over it._
* Provide a simple example on how to use the overlay especially when transitioning from the kernel module.  _Examples bring clarity._

closed 2022-02-13 by github-actions[bot]