notro/fbtft #558: sainsmart18.dts can not get it to run on new 5.4 kernel

rudiratlos, 2020-07-22:
I couldn't get this to work. My display is a sainsmart18 with st7735r controller.
With fbtft_device before new 5.4 kernel this works great. I tried to compose a .dts file, but this doesnt work.
Any idea?
 
`// Definitions for sainsmart1.8 display
// dtc -@ -I dts -O dtb -o /boot/overlays/sainsmart18.dtbo sainsmart18.dts
// add entry to /boot/config.txt: dtoverlay=sainsmart18
/dts-v1/;
/plugin/;

/ {
  compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

  fragment@0 {
    target = <&spi0>;
    __overlay__ {
      status = "okay";
      spidev@0 {
          status = "disabled";
      };
      spidev@1 {
          status = "disabled";
      };
    };
  };

  fragment@1 {
    target = <&gpio>;
    __overlay__ {
      sainsmart18_pins: sainsmart18_pins {
          brcm,pins = <27 9>;
          brcm,function = <1 1>;
      };
    };
  };

  fragment@2 {
    target = <&spi0>;
    __overlay__ {
      /* needed to avoid dtc warning */
      #address-cells = <1>;
      #size-cells = <0>;

      sainsmart18: sainsmart18@0 {
        compatible = "sitronix,sainsmart18";
        reg = <0>;
        pinctrl-names = "default";
        pinctrl-0 = <&sainsmart18_pins>;

        spi-max-frequency = <16000000>;
        txbuflen = <32768>;
        bgr = <0>;
        fps = <30>;
        buswidth = <8>;

        reset-gpios = <&gpio 27 0>;
        dc-gpios = <&gpio 9 0>;
        debug = <0>;
        
        //init = <0x10000ae /* Display Off */
        //        0x10000a0 0x72 /* Set Colour Depth and left to right */
        //        0x10000a1 0x00 /* Set Display Start Line */
        //        0x10000a2 0x00 /* Set Display Offset */
        //        0x10000a4 /* Normal Display */
        //        0x10000a8 0x3f /* Set multiplex */
        //        0x10000ad 0x8e /* Set master */
        //        /*0x10000b0 0x0b  Set power mode */
        //        0x10000b1 0x31 /* Precharge */
        //        0x10000b3 0xf0 /* Clock Div */
        //        0x100008a 0x64 /* Precharge A */
        //        0x100008b 0x78 /* Precharge B */
        //        0x100008c 0x64 /* Precharge C  */
        //        0x10000bb 0x3a /* Precharge level */
        //        0x10000be 0x3e /* vcomh */
        //        0x1000087 0x06 /* Master current */
        //        0x1000081 0x91 /* Contrast A */
        //        0x1000082 0x50 /* Contrast B */
        //        0x1000083 0x7d /* Contrast C */
        //        0x10000af /* Set Sleep Mode Display On */
        //        >;
        };
      };
    };
    __overrides__ {
            speed =	<&sainsmart18>,"spi-max-frequency:0";
            txbuflen =	<&sainsmart18>,"txbuflen:0";
            fps =	<&sainsmart18>,"fps:0";
            bgr =	<&sainsmart18>,"bgr:0";
            debug =	<&sainsmart18>,"debug:0";
    };
};
`

rudiratlos, 2020-07-22:
I'm a little bit further, but still not working. I think I have problems with assiging the dc-pin.

[https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=280676&p=1700677#p1700677](url)

hackerjimbo, 2020-07-27:
Please keep up the good work! I'm having similar difficulties and haven't got _anywhere_!

F1zye, 2020-08-12:
The following works for me, when setting pwm on commandline with `gpio -g mode 18 pwm && gpio -g pwm 18 100`.
(PWM in the dts didn't work for me)
References:
* https://lists.freedesktop.org/archives/dri-devel/2017-December/159939.html
* https://www.kernel.org/doc/Documentation/devicetree/bindings/display/sitronix%2Cst7735r.txt
* https://www.kernel.org/doc/Documentation/devicetree/bindings/display/sitronix%2Cst7735r.yaml
* https://gist.github.com/pdp7/ab0344e4f0d6348e3ad3a7f834cbd3ca

`*
 * Device Tree overlay for st7735r
 *
 */

/dts-v1/;
/plugin/;

/ {
        compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

        fragment@0 {
                target = <&spi0>;
                __overlay__ {
                        status = "okay";
                };
        };

        fragment@1 {
                target = <&spidev0>;
                __overlay__ {
                        status = "disabled";
                };
        };
/*
        fragment@2 {
                target = <&spidev1>;
                __overlay__ {
                        status = "disabled";
                };
        };
*/
        fragment@3 {
                target = <&gpio>;
                __overlay__ {
                        st7735r_display_pins: st7735r_display_pins {
                                brcm,pins = <24 25>;
                                brcm,function = <1 1>;
                                brcm,pull = <0 0>;
                        };
                };
        };
/*      fragment@4 {
                target-path = "/";
                __overlay__ {
                        backlight_gpio: backlight_gpio {
                                compatible = "gpio-backlight";
                                gpios = <&gpio 18 1>;
                        };

                        backlight_pwm: backlight_pwm {
                                pinctrl-names = "default";
                                pinctrl-0 = <&backlight_pwm_pins>;
                                status = "okay";
                                compatible = "pwm-backlight";
                                pwms = <&ehrpwm1 1 500000 0>;
                                brightness-levels = <
                                        0  1  2  3  4  5  6  7  8  9
                                        10 11 12 13 14 15 16 17 18 19
                                        20 21 22 23 24 25 26 27 28 29
                                        30 31 32 33 34 35 36 37 38 39
                                        40 41 42 43 44 45 46 47 48 49
                                        50 51 52 53 54 55 56 57 58 59
                                        60 61 62 63 64 65 66 67 68 69
                                        70 71 72 73 74 75 76 77 78 79
                                        80 81 82 83 84 85 86 87 88 89
                                        90 91 92 93 94 95 96 97 98 99
                                        100
                                >;
                                default-brightness-level = <90>;
                        };
                };
        };*/

        fragment@5 {
                target = <&spi0>;
                __overlay__ {
                        /* needed to avoid dtc warning */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        
                        st7735rdisplay: st7735r-display@0{
                                compatible = "jianda,jd-t18003-t01", "sitronix,st7735r";
                                reg = <0>;
                                pinctrl-names = "default";
                                pinctrl-0 = <&st7735r_display_pins>;

                                spi-max-frequency = <32000000>;
                                rotation = <90>;
                                //bgr = <0>;
                                fps = <20>;
                                height = <160>;
                                width = <128>;
                                buswidth = <8>;
                                reset-gpios = <&gpio 25 0>;
                                dc-gpios = <&gpio 24 0>;
                                //led-gpios = <&gpio 18 0>;
                                //backlight = &backlight_gpio;
                                //debug = <0>;
                        };

                };
        };
        __overrides__ {
                speed =     <&st7735rdisplay>,"spi-max-frequency:0";
                rotate =    <&st7735rdisplay>,"rotation:0";
                fps =       <&st7735rdisplay>,"fps:0";
                debug =     <&st7735rdisplay>,"debug:0";
        };
};
`

rudiratlos, 2020-08-19:
this compiles and loads on my rpi. can someone be so kind to test it. my display broke down.
It can modify pin assignments (psl. see config.txt entry)

`/*
 * Device Tree overlay for st7735r based SPI OLED display
 *
 * /boot/config.txt
 * dtoverlay=st7735r,dc_pin=13,reset_pin=27,debug=3,bgr=1
 *
 * compile:
 * dtc -@ -I dts -O dtb -o /boot/overlays/st7735r.dtbo st7735r.dts
 */

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835";

	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@1 {
		target = <&spidev0>;
		__overlay__ {
			status = "disabled";
		};
	};

	fragment@2 {
		target = <&spidev1>;
		__overlay__ {
			status = "disabled";
		};
	};

	fragment@3 {
		target = <&gpio>;
		__overlay__ {
			st7735_pins: st7735_pins {
                                brcm,pins = <24 25>;
                                brcm,function = <1 1>; /* out out */
			};
		};
	};

	fragment@4 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			st7735: st7735@0{
				compatible = "sitronix,st7735r", "sitronix,sainsmart18";
				reg = <0>;
				pinctrl-names = "default";
				pinctrl-0 = <&st7735_pins>;

				spi-max-frequency = <32000000>;
				bgr = <0>;
				bpp = <1>;
				rotate = <0>;
				fps = <25>;
				buswidth = <8>;
				reset-gpios = <&gpio 25 0>;
				dc-gpios = <&gpio 24 0>;
				debug = <0>;

				sitronix,height = <128>;
				sitronix,width = <160>;
				sitronix,page-offset = <0>;
			};
		};
	};

	__overrides__ {
		speed     = <&st7735>,"spi-max-frequency:0";
		rotate    = <&st7735>,"rotate:0";
		fps       = <&st7735>,"fps:0";
		bgr       = <&st7735>,"bgr:0";
		debug     = <&st7735>,"debug:0";
		dc_pin    = <&st7735>,"dc-gpios:4",
		            <&st7735_pins>,"brcm,pins:4";
		reset_pin = <&st7735>,"reset-gpios:4",
		            <&st7735_pins>,"brcm,pins:0";
		height    = <&st7735>,"sitronix,height:0";
		width     = <&st7735>,"sitronix,width:0";
	};
};`

closed 2021-08-24 by github-actions[bot]