notro/fbtft #560: can not get adafruit_green display to run correctly on 5.x kernel

rudiratlos, 2020-08-24:
I've figuered out, that adafruit_green is nearly compatible with sainsmart18 (st7735r driver).
But adafruit_green buffer will start 2 colums further, compared to sainsmart18. 
That's why we will see pixel artefacts in top row and most left column of the adafruit display.

In the old module fbtft_device.c (not available in new kernel) there is a specific 
"adafruit18_green_tab_set_addr_win" function, which will cope with this 2 column offset. But unfortunatly fbtft_device.c will not be used anymore in new kernel. So there is no specfic setup for the adafruit display and we will see these ugly pixel artefacts.

How can I feed "set_addr_win" function with specifc adafruit display code??

code snippets, which will show the problem:

`In MIPI_DCS_SET_COLUMN_ADDRESS	= 0x2A,
MIPI_DCS_SET_PAGE_ADDRESS	= 0x2B,

im linux staging code in st7735r.c als default
static void set_addr_win(struct fbtft_par *par, int xs, int ys, int xe, int ye)
{
	write_reg(par, MIPI_DCS_SET_COLUMN_ADDRESS,
		  xs >> 8, xs & 0xFF, xe >> 8, xe & 0xFF);

	write_reg(par, MIPI_DCS_SET_PAGE_ADDRESS,
		  ys >> 8, ys & 0xFF, ye >> 8, ye & 0xFF);

	write_reg(par, MIPI_DCS_WRITE_MEMORY_START);
}

adafruit_green specific code (fbtft_device.c is not available in new kernel):
static void adafruit18_green_tab_set_addr_win(struct fbtft_par *par,
						int xs, int ys, int xe, int ye)
{
	fbtft_par_dbg(DEBUG_SET_ADDR_WIN, par,
		"%s(xs=%d, ys=%d, xe=%d, ye=%d)\n", __func__, xs, ys, xe, ye);
	write_reg(par, 0x2A, 0, xs + 2, 0, xe + 2);
	write_reg(par, 0x2B, 0, ys + 1, 0, ye + 1);
	write_reg(par, 0x2C);
	
	.bgr = true,
}`

notro, 2020-08-31:
Ref: https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=281204

notro, 2020-08-31:
There are 2 solutions to this:

1. Add support to [drivers/staging/fbtft/fb_st7735r.c](https://elixir.bootlin.com/linux/latest/source/drivers/staging/fbtft/fb_st7735r.c)  
  This can be done by adding a ```struct fbtft_display``` and open coding ```FBTFT_REGISTER_DRIVER``` in order to add support for a second DT compatible string.

2. Add support to [drivers/gpu/drm/tiny/st7735r.c](https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/tiny/st7735r.c)   
See [drm: Add support for Okaya RH128128T](https://patchwork.freedesktop.org/series/71567/) for how a similar display with pixel offsets was supported. Long term all fbtft drivers have to move to drm.

Restoring fbtft_device will not bring back support because it was already broken due to a gpio refactoring that broke things left and right.

fbtft doesn't have a maintainer that looks after it, and unfortunately I can't help you with this.

inkyusa, 2020-09-17:
I guess this issue already resolved in https://github.com/raspberrypi/linux/commit/10de25851e6850e734f7655df9deea1eaedf0019.
Haven't tested yet..

rudiratlos, 2020-09-21:
my display works now. but still have problems with bgr. red color appears blue and blue appears red. the parameter bgr is not working

rudiratlos, 2020-09-21:
pls. see:
`https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=281204&start=25#p1730283`

closed 2021-08-24 by github-actions[bot]