notro/fbtft #563: TFT_320QVT via SPI problem

radekczdev, 2020-12-03:
Hi,
I am trying to connect TFT_320QVT using sainsmart32_spi. The problem is, that after initializing the screen and running startx, screen looks like on below photos (moving mouse causes change of the colors/vertical lines):
[
[image]
[image]
[image]

](url)

I have connected it using the SPI circuit from WIKI. The only (maybe big?) difference is that I don't have 74HC04 nor 74HCT04. I am using a circuit with NPN 2N2222A transistor. After checking the signal on IN/ON when moving mouse, when input is changing 0-1, the output seems not keep up with the changes and it holds 0 or 1 during the input switching.
May it be the problem? Or it might be some misconfiguration?

In addition, after initialization, I connect the in/out WR directly (omitting the NOT gate), the screen works like on below photo. When moving the mouse, the screen is "alive" - literally. The images are overlaying, the mouse is dissapearing.
[image]

Do you have any thoughts on this one? May it be the 2N2222A problem? Or other circuits like 74HC4040 or 74HC4094? I am powering all the stuff using 3.3V from RPI. The power supply I have for RPI provides output current 2.1A.

Best Regards,
Radek

CRImier, 2020-12-03:
Hello! You should try using a buffer instead of the 2N2222, as the output signal might not switch fast enough when you switch the 2N2222 state. What's your circuit? If you really can't get any inverters, try decreasing the pullup on the 2N2222 output to, say, 1K - I assume it's larger than that.

radekczdev, 2020-12-03:
@CRImier Actually it is 1k at the output and 15k on the input. Vcc 3,3V.

CRImier, 2020-12-03:
Perhaps decrease the base (input) resistor to 1k, too? A schematic would be cool to see, either way.

radekczdev, 2020-12-04:
I have tried to decrease the base and emitter to 470 both, but it didn't help - still signal on the output cannot keep up changes on base. I am waiting for the inverters. I will update the status and describe what was wrong after I will try them - for the others, so they won't need to look for the solution as I do :)

radekczdev, 2020-12-07:
I am using https://github.com/notro/fbtft/wiki/SPI-interface-circuit circuit. I have connected the HC04 now. What happens is that screen is being initialized (to black one). After starting startx it is fading back to white.. I am using below script to start (maybe there is a wrong sequence in initialization?
`#!/bin/bash
sudo modprobe spi-bcm2835;
wait
sudo rmmod fb_ssd1289;
wait
sudo rmmod fbtft_device; 
wait
sudo rmmod fbtft;
wait
#sudo modprobe --first-time fbtft dma=false;
wait
sudo modprobe fbtft_device name=sainsmart32_spi speed=16000000 debug=$((1<<5)) gpios=reset:25,dc:24 fps=19 rotate=90 width=240 bgr=0 debug=3 height=320; #  buswidth=16; # txbuflen=32768;
#wait
#sudo mv /usr/share/X11/xorg.conf.d/{,.}99-fbturbo.conf;
#wait
#sudo FRAMEBUFFER=/dev/fb1 startx #-- -dpi 60

`

radekczdev, 2020-12-08:
I will just add that con2fbmap 1 1 does nothing.

closed 2021-08-24 by github-actions[bot]