notro/fbtft #564: st7789v kernel 5.4 color error

DRAgon734465502, 2020-12-15:
I use this 1.54inch LCD  240x240 st7789v
https://www.aliexpress.com/item/3297003 ... 2970035492

But at kernel 5.4 the color not right 
like this 
[https://ae01.alicdn.com/kf/H15c23d5a1f2d44d0865c1439f183cc50v.png](url)

I do a dts , code like this 

`                                spi-max-frequency = <48000000>;
                                rotate = <270>;
                                width = <240>;
                                height = <240>;
                                fps = <30>;
                                buswidth = <8>;
                                txbuflen = <32768>;
                                reset-gpios = <&gpio 27 1>;
                                dc-gpios = <&gpio 22 0>;
                                spi-cpol;
                                spi-cpha;
                                bgr;
                                debug = <0>;
                                init = <0x1000011
                                        0x20000ff
                                        0x100003A 0x55
                                        0x10000C0 0x0C
                                        0x1000021
                                        0x1000029   >;`

DRAgon734465502, 2020-12-15:
I do not know why  the yellow color Seems more brighter at kernel 5.4

DRAgon734465502, 2020-12-15:
the color is more black at kernel 5.4 

https://ae01.alicdn.com/kf/H6f5572e54f7547c08d10f6ae377ded28W.png

breaktime1903, 2021-02-06:
I want to ask how to use the st7789v LCD Display in Armbian.I need it just now!

AVTAR0901, 2021-03-22:
Hey @DRAgon734465502 
How did you get Raspbian buster working with ST7789v 240x240 I have similar display but its not mirroring properly. can u help me with any link or something that made it work for you?? it will be great help.
I was following below link to get it work.
https://krystof.io/mirroring-raspberry-pi-hdmi-video-to-a-st7789-1-3-inch-lcd-display/

closed 2021-08-24 by github-actions[bot]