notro/fbtft #567: I cant use fbtft_device

breaktime1903, 2021-02-06:
I found some error when I was using st7789 LCD Display.

I can't insert the module called 'fbtft_device' into the kernel.And I didn't found the module in /lib/modules.

Please help me!!!

breaktime1903, 2021-02-06:
I know the module has been removed after Linux 5.4,but I need it!

modi12jin, 2021-02-08:
@breaktime1903 

[image]

I am not sure how to do it.

breaktime1903, 2021-02-18:
你也遇到这个问题了？
我特喵搞了半天，就只见到fbtft.ko

modi12jin, 2021-02-18:
@breaktime1903 If you are using Raspberry Pi, you can refer to this
> https://github.com/juj/fbcp-ili9341

modi12jin, 2021-02-22:
@breaktime1903  https://b23.tv/TGFsDn

Jpe230, 2021-02-27:
> @breaktime1903  https://b23.tv/TGFsDn

Interesting findings, I am driving an unknown ST7789V, while I was able to initialize it using fbtft and a custom overlay, I couldn't get the correct gamma nor fix the offset, even though I'm running a initialization code that I know for sure works in pre 5.4

I wonder what kind of problems did the author encounter with fbtft.

For now 4.9 is making the job but it is sure that the driver in the kernel is totally borked.

modi12jin, 2021-04-12:
@breaktime1903 
https://devdotnet.org/post/rabota-s-gpio-v-linux-na-primere-banana-pi-bpi-m64-chast-4-device-tree-overlays-podkluchenie-displey-spi-lcd-ili9341/

Great, I found the answer!

closed 2021-08-24 by github-actions[bot]