notro/fbtft #569: @breaktime1903

breaktime1903, 2021-02-18:
@breaktime1903 

[image]

I am not sure how to do it.

_Originally posted by @modi12jin in https://github.com/notro/fbtft/issues/567#issuecomment-774871435_

breaktime1903, 2021-02-18:
这貌似要改DTS了

breaktime1903, 2021-02-18:
谁能帮我一下

modi12jin, 2021-04-14:
@breaktime1903 

[image]

[image]

[image]

> sun8i-h3-spi-st7789v-led-always-on.dts

```
/dts-v1/;
/plugin/;

/ {
    compatible = "allwinner,sun8i-h3";    

  fragment@0 {
    target = <&pio>;
    __overlay__ {
      st7789v_pins: st7789v_pins {
        pins = "PG7", "PG6"; /*RESET, DC_RS*/
        function = "gpio_out",;
      };
    };
  };
  
  fragment@1 {
    target = <&spi1>;
    __overlay__ {
      #address-cells = <1>;
      #size-cells = <0>;
      status = "okay";      
      cs-gpios = <&pio 0 16 0>; /* PA16 */

      st7789v: st7789v@0 {
        compatible = "sitronix,st7789v";
        reg = <0>;
        pinctrl-names = "default";
        pinctrl-0 = <&st7789v_pins>;
        spi-max-frequency = <16000000>;
        rotate = <90>;
        bgr;
        fps = <30>;
        height = <240>;
        width = <240>;
        buswidth = <8>;
        reset-gpios = <&pio 6 7 1>; /*RESET=PG7*/
        dc-gpios = <&pio 6 6 0>; /*DC_RS=PG6*/
        /*led-gpios = <&pio 2 4 0>; LED=PC4*/
        debug = <0>;
        init = <0x1000011
        0x20000ff
        0x100003A 0x55
        0x10000C0 0x0C
        0x1000021
        0x1000029 >;
      };
    };
  };
};
```

```
dtc -O dtb -o sun8i-h3-spi-st7789v-led-always-on.dtbo sun8i-h3-spi-st7789v-led-always-on.dts
```

modi12jin, 2021-04-14:
> @ breaktime1903
>
> [image]
>
> [image]
>
> [image]
>
>
> ```
> /dts-v1/;
> /plugin/;
>
> / {
>     compatible = "allwinner,sun8i-h3";
>
>   fragment@0 {
>     target = <&pio>;
>     __overlay__ {
>       st7789v_pins: [...]

I don't know how to change to spi1

modi12jin, 2021-04-14:
[image]

modi12jin, 2021-04-15:
[image]

```
overlays=spi-spidev
param_spidev_spi_bus=1
param_spidev_spi_cs=1
```

[image]

```
dmesg | grep spi
```

modi12jin, 2021-04-26:
@breaktime1903 

[image]

It is estimated that there is a problem with initialization

modi12jin, 2021-04-26:
@breaktime1903 

[image]

[image]

https://user-images.githubusercontent.com/40233017/116118087-8d1fd400-a6ef-11eb-942e-5a24aa64dc1f.mp4

closed 2021-08-24 by github-actions[bot]