notro/fbtft #571: rasbian mz61581 no '/sys/class/backlight/' anymore

mojocorp, 2021-03-06:
Hi,

Since I've updated raspbian recently, the '/sys/class/backlight/' is completely gone. I cannot control the backlight anymore using 'echo 1 | sudo tee /sys/class/backlight/*/bl_power'.
I'm still able to control the backlight using python on gpio 12:

# turn off
GPIO.output(12, GPIO.HIGH)
# turn off
GPIO.output(12, GPIO.LOW)

Since I did not update since years I may missed few things....

ahogen, 2021-07-06:
Per README:

> 2015-01-19
> The FBTFT drivers are now in the Linux kernel staging tree: https://git.kernel.org/cgit/linux/kernel/git/gregkh/staging.git/tree/drivers/staging/fbtft?h=staging-testing
>
> ### **Development in this github repo has ceased.**

peterlin82, 2021-09-09:
I have the same problem.
My LCD is ili9341,rpi-display.dtbo,2021-01-11-raspios-buster-armhf-lite.zip(kernel 5.4) or 2021-05-07-raspios-buster-armhf-lite.zip(kernel 5.10)

**### /sys/class/backlight/ is empty**

I don't know how to solve.

notro, 2021-11-08:
The problem is fixed here: [[PATCH v2] staging/fbtft: Fix backlight](https://lore.kernel.org/dri-devel/20211105204358.2991-1-noralf@tronnes.org/)
This will eventually trickle back through the kernel stable tree.

notro, 2021-12-01:
The fix is now in [mainline](https://github.com/torvalds/linux/commit/7865dd24934ad580d1bcde8f63c39f324211a23b) and in the stable releases 5.15.6, 5.10.83 and 5.4.163.

closed 2022-02-07 by github-actions[bot]