notro/fbtft #576: spi ili9341 screen black after update to 5.10.60-sunxi ( fb fbtft )

naty6458, 2021-12-21:
​
Hello, 

 

i update my OrangePiZero OS from Linux 5.10.43-sunxi to Linux 5.10.60-sunxi
my screen fbtft spi lcd stop working , the screen is black.

If there is a solution to this issue I would love to get help, thanks!
here my configuration (work on 5.10.43).

 

fbtft.dts

 

/dts-v1/;
/plugin/;

/ {
	compatible = "allwinner,sun8i-h3";

	fragment@0 {
		target = <&spi1>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@1 {
		target = <&pio>;
		__overlay__ {
			spi1_cs_pins: spi1_cs_pins {
					pins = "PA13";
					function = "gpio_out";
			};

			opiz_display_pins: opiz_display_pins {
                pins = "PA7", "PA2", "PA6";
                function = "gpio_out";
			};
		};
	};

	fragment@2 {
		target = <&spi1>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;
			pinctrl-0=<&spi1_pins &spi1_cs_pins>;
			cs-gpios=<&pio 0 13 1>;

			opizdisplay: opiz-display@0{
				compatible = "ilitek,ili9341";
				reg = <0>;
				pinctrl-names = "default";
				pinctrl-0 = <&opiz_display_pins>;
				spi-max-frequency = <32000000>;
				rotate = <270>;
				bgr;
				fps = <30>;
				buswidth = <8>;
				reset-gpios = <&pio 0 7 1>;
				dc-gpios = <&pio 0 2 0>;
				led-gpios= <&pio 0 6 0>;
				debug = <5>;
				status="okay";
			};
		};
	};
	__overrides__ {
		speed =   <&opizdisplay>,"spi-max-frequency:0";
		rotate =  <&opizdisplay>,"rotate:0";
		fps =     <&opizdisplay>,"fps:0";
		debug =   <&opizdisplay>,"debug:0";
        };
};

 

/boot/armbianEnv.txt  is:

verbosity=1
bootlogo=true
console=serial
overlay_prefix=sun8i-h3
overlays=analog-codec i2c0 uart1 uart2
user_overlays=fbtft mcp7940
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u

​

notro, 2021-12-21:
> i update my OrangePiZero OS from Linux 5.10.43-sunxi to Linux 5.10.60-sunxi
my screen fbtft spi lcd stop working , the screen is black.

fbtft history:
* [v5.10.43](https://github.com/gregkh/linux/commits/v5.10.43/drivers/staging/fbtft)
* [v5.10.60](https://github.com/gregkh/linux/commits/v5.10.60/drivers/staging/fbtft)

This patch is new in 5.10.60 and looks like it changes the reset gpio polarity:
[staging: fbtft: Rectify GPIO handling](https://github.com/gregkh/linux/commit/7a42f3c30dfe3d51fa7d1a70b669243d9b34a38d)

Also backlight control has been broken since 5.4, fixed in 5.10.83: #571

naty6458, 2021-12-22:
Hi First of all thank you very much.
I finally managed to turn on the screen,
The problem was, in the polarization of the LED pin (backlight).
Before that it was on 0 when I switched it to 1 everything suddenly works.
The connection to the screen worked fine, but there was no lighting to display the content :-)

The DTS file that worked for me is: (ORANGE PI ZERO) 

reset ---> PA7
led ---> PA6
dc ----> PA2

```
/dts-v1/;
/plugin/;

/ {
	compatible = "allwinner,sun8i-h3";

    fragment@0 {
        target-path = "/aliases";
        __overlay__ {
            spi1 = "/soc/spi@01c69000";
        };
    };

	fragment@1 {
		target = <&spi1>;
		__overlay__ {
			status = "okay";

			spidev@0{
				status = "disabled";
			};

			spidev@1{
				status = "disabled";
			};
		};
	};

	fragment@2 {
		target = <&pio>;
		__overlay__ {
			opiz_display_pins: opiz_display_pins {
                pins = "PA7", "PA2", "PA6";
                function = "gpio_out";
			};
		};
	};

	fragment@3 {
		target = <&spi1>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;
			opizdisplay: opiz-display@0{
				compatible = "ilitek,ili9341";
				reg = <0>;
				pinctrl-names = "default";
				pinctrl-0 = <&opiz_display_pins>;
				spi-max-frequency = <8000000>;
				rotate = <270>;
				bgr;
				fps = <30>;
				buswidth = <8>;
				reset-gpios = <&pio 0 7 1>;
				dc-gpios = <&pio 0 2 0>;
				led-gpios= <&pio 0 6 1>;
				debug = <3>;
				status="okay";
			};
		};
	};
	__overrides__ {
		speed =   <&opizdisplay>,"spi-max-frequency:0";
		rotate =  <&opizdisplay>,"rotate:0";
		fps =     <&opizdisplay>,"fps:0";
		debug =   <&opizdisplay>,"debug:0";
        };
};

```
[image]

closed 2022-02-28 by github-actions[bot]