notro/fbtft #577: HardKernel Odroid OGST TFT on Android

klandrith, 2022-01-05:
Hello,

I have an Odroid N2 made by HardKernel. I am using an OGST Gaming Case (https://ameridroid.com/products/ogst-gaming-console-case-for-odroid-xu4) that is designed for an Odroid XU4. I am trying to get that built in 2.4" TFT to work. Now here is some basic background info. I am running a custom LineageOS 18.1 image made by a gentleman over on the Odroid forums by the name of voodik. Under /dev/graphics there are 4 framebuffer devices listed. Their resolution as reported by 

`cat /sys/class/graphics/fb*/virtual_size` 

is..

1920, 2160
32, 32
32, 32
32, 32

The reason that I mention these devices is that I have no idea what fb1 - 3 refer to.

Now, when I enter the following command over adb, 

`modprobe -d /vendor/lib/modules fbtft_device name=hktft9340 busnum=0 rotate=270`

I get the following dmesg output...

[  212.921957] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[  212.923830] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[  212.924264] spidev spi0.0: spidev spi0.0 100000kHz 8 bits mode=0x00
[  212.924310] meson-fb meson-fb: meson-fb id=-1 pdata? no
[  212.924343] spidev spi0.0: Deleting spi0.0
[  212.924632] spi spi0.0: setup mode 0, 8 bits/w, 40000000 Hz max --> 0
[  212.924747] meson-spicc ffd13000.spi: registered child spi0.0
[  212.924750] fbtft_device: GPIOS used by 'hktft9340':
[  212.924752] fbtft_device: 'reset' = GPIO480
[  212.924754] fbtft_device: 'dc' = GPIO483
[  212.924755] fbtft_device: 'led' = GPIO473
[  212.924759] spi spi0.0: fb_ili9340 spi0.0 40000kHz 8 bits mode=0x00
[  214.249383] ffs_retry_malloc_buffer

As can be seen, there is never an additional framebuffer device that is ceated... 

`odroidn2:/dev/graphics # ls /dev/graphics`
fb0  fb1  fb2  fb3

...and the fb1-3 devices report the same display size, so fbtft is not binding to one of them.

I haven't been able to readily find any real info on doing this under Android. Odroid had this working under Android with their XU4 boards, but I haven't been able to find any real details on how they got that working. Any ideas?

notro, 2022-01-05:
The task of [fbtft_device](https://elixir.bootlin.com/linux/v4.14/source/drivers/staging/fbtft/fbtft_device.c) is to add a device describing the display. In your case this is the device added:
 ```
[ 212.924759] spi spi0.0: fb_ili9340 spi0.0 40000kHz 8 bits mode=0x00
 ```
On most systems the fb_ili9340 module/driver would be loaded automatically by udev when the device is added and the driver would register the framebuffer.
You could try loading the module manually: ```modprobe -d /vendor/lib/modules fb_ili9340```

klandrith, 2022-01-05:
Thanks for the help! That did the trick and created a framebuffer device and initiated the display with the backlight coming on. I haven't gotten it to display anything yet though. I'm attempting to compile a static version of mplayer for my purposes.
Thanks again!

klandrith, 2022-01-05:
I seem to be still having a problem, at least according to what you posted in another issue thread. As soon as the framebuffer device is created after issuing `modprobe -d /vendor/lib/modules fb_ili9340` the screen goes blank white with the backlight on. It is my understanding that it is supposed to go blank black? Correct?

klandrith, 2022-01-05:
So, I'm unsure how to do this from the command line calling fbtft_device via modprobe. The following are the default pins for my board...

[  781.757402] fbtft_device: 'reset' = GPIO480
[  781.757404] fbtft_device: 'dc' = GPIO483
[  781.757405] fbtft_device: 'led' = GPIO473

However, pins 480 and 483 are active low, while 473 is active high. I do not know if the hktft9340 module "knows" this or not. If it doesn't, I would presume that is my problem with the display not initializing correctly and the reason for the blank white screen after fb_ili9340 loading. How can I specify the pin state when I load fbtft_device via modprobe. I know to specify the pin is gpios=reset:480....etc...

notro, 2022-01-06:
> the screen goes blank white with the backlight on.

This is a typical controller that has not been initialized correctly.

> How can I specify the pin state when I load fbtft_device via modprobe.

fbtft_device can't do that. It's possible when adding the device through Device Tree, but things have changed over the years and I'm not sure about this, it might depend on the kernel version.

reset should be active high in [old](https://elixir.bootlin.com/linux/v4.14/C/ident/fbtft_reset) fbtft and active low since Linux 5.3: https://github.com/torvalds/linux/commit/b918d1c2706619cb0712a61cc8c05148b68b24b2
dc and led should be active high.

klandrith, 2022-01-08:
Ok...I've double and triple checked my wiring and labeled everything and re-hooked it all up. I am loading the following....

modprobe -d /vendor/lib/modules fbtft_device name=hktft9340 busnum=0 rotate=270 gpios=reset:472,led:473,dc:482
modprobe -d /vendor/lib/modules fb_ili9340

Now before I load the modules, `gpio readall` reports the following...

I/O 484 - wPi 12 - MOSI - ALT4 - 1
I/O 485 - wPi 13 - MISO - ALT4 - 1
I/O 487 - wPi 14 - SCLK - ALT4 - 1
I/O 486 - wPi 10 - CE0   - OUT - 1

I/O 472 - wPi 26 - IO.472 - IN - 0   (Reset)
I/O 473 - wPi 7   - IO.473 - IN - 0   (LED)
I/O 482 - wPi 24 - IO.482 - IN - 0   (DC)

And after loading the above modules with those parameters I get...

I/O 484 - wPi 12 - MOSI - ALT4 - 0
I/O 485 - wPi 13 - MISO - ALT4 - 1
I/O 487 - wPi 14 - SCLK - ALT4 - 0
I/O 486 - wPi 10 - CE0   - OUT - 1

I/O 472 - wPi 26 - IO.472 - IN - 1   (Reset)
I/O 473 - wPi 7   - IO.473 - IN - 1   (LED)
I/O 482 - wPi 24 - IO.482 - IN - 1   (DC)

So, that still gets me a blank back lit white screen. So, I got a hold of the HardKernel dtbo file and initialized fbtft_device with the values and arguments present in it.
`modprobe -d /vendor/lib/modules fbtft_device name=hktft9340 reg=0 pinctrl-names="default" pinctrl-0=1 speed=32000000 busnum=0 rotate=270 bgr=1 backlight=1 fps=20 bpp=16 buswidth=8 gpios=reset:472,led:473,dc:482 debug=7`

Followed by the same `modprobe -d /vendor/lib/modules fb_ili9340`

Which yielded me the same thing, and the following dmesg output.

[  234.541364] fbtft_device: unknown parameter 'reg' ignored
[  234.541366] fbtft_device: unknown parameter 'pinctrl-names' ignored
[  234.541368] fbtft_device: unknown parameter 'pinctrl-0' ignored
[  234.541374] fbtft_device: unknown parameter 'backlight' ignored
[  234.541376] fbtft_device: unknown parameter 'bpp' ignored
[  234.541566] spidev spi0.0: spidev spi0.0 100000kHz 8 bits mode=0x00
[  234.541614] meson-fb meson-fb: meson-fb id=-1 pdata? no
[  234.541648] spidev spi0.0: Deleting spi0.0
[  234.541925] spi spi0.0: setup mode 0, 8 bits/w, 32000000 Hz max --> 0
[  234.542041] meson-spicc ffd13000.spi: registered child spi0.0
[  234.542045] fbtft_device: GPIOS used by 'hktft9340':
[  234.542046] fbtft_device: 'reset' = GPIO472
[  234.542048] fbtft_device: 'led' = GPIO473
[  234.542049] fbtft_device: 'dc' = GPIO482
[  234.542053] spi spi0.0: fb_ili9340 spi0.0 32000kHz 8 bits mode=0x00
[  245.375328] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[  245.376168] fb_ili9340: module is from the staging directory, the quality is unknown, you have been warned.
[  245.376596] fb_ili9340 spi0.0: fbtft_request_gpios_match('reset')
[  245.376613] fb_ili9340 spi0.0: fbtft_request_gpios: 'reset' = GPIO472
[  245.376615] fb_ili9340 spi0.0: fbtft_request_gpios_match('led')
[  245.376621] fb_ili9340 spi0.0: fbtft_request_gpios: 'led' = GPIO473
[  245.376623] fb_ili9340 spi0.0: fbtft_request_gpios_match('dc')
[  245.376629] fb_ili9340 spi0.0: fbtft_request_gpios: 'dc' = GPIO482
[  245.376631] fb_ili9340 spi0.0: fbtft_verify_gpios()
[  245.376633] fb_ili9340 spi0.0: fbtft_reset()
[  245.496665] fb_ili9340 spi0.0: fbtft_write_reg8_bus8: ef 03 80 02
[  245.496671] fb_ili9340 spi0.0: fbtft_write_spi(len=1): ef
[  245.502050] fb_ili9340 spi0.0: fbtft_write_spi(len=3): 03 80 02
[  245.502079] fb_ili9340 spi0.0: fbtft_write_reg8_bus8: cf 00 c1 30
[  245.502083] fb_ili9340 spi0.0: fbtft_write_spi(len=1): cf
[  245.502105] fb_ili9340 spi0.0: fbtft_write_spi(len=3): 00 c1 30
[  245.502127] fb_ili9340 spi0.0: fbtft_write_reg8_bus8: ed 64 03 12 81
[  245.502131] fb_ili9340 spi0.0: fbtft_write_spi(len=1): ed
[  245.502152] fb_ili9340 spi0.0: fbtft_write_spi(len=4): 64 03 12 81
[  245.502174] fb_ili9340 spi0.0: fbtft_write_reg8_bus8: e8 85 00 78
[  245.502177] fb_ili9340 spi0.0: fbtft_write_spi(len=1): e8
[... 83 lines cut ...]
[  245.696861] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  245.700061] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  245.703252] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  245.706453] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  245.709727] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  245.712935] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  245.716138] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  245.719306] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  245.722519] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  245.725787] fb_ili9340 spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[... cut ...]
[  245.749322] fb_ili9340 spi0.0: fbtft_backlight_update_status: polarity=1, power=0, fb_blank=0
[  260.731800] ffs_retry_malloc_buffer

Then I figured I would do some research on the Amlogic SPICC controller module. I found that it only supports up to 30mhz, so I tried dropping the bus speed down to 30mhz, then 20mhz, and then 10mhz, and got the same result as well. Not quite sure what to try now.

BTW: This build of android is using kernel 4.9, so I assume the led output should be active high per your previous comment.

Thanks again for the help.

notro, 2022-01-08:
> I/O 472 - wPi 26 - IO.472 - IN - 1 (Reset)
> I/O 473 - wPi 7 - IO.473 - IN - 1 (LED)
> I/O 482 - wPi 24 - IO.482 - IN - 1 (DC)

This doesn't look right, IN?

You could try and see if backlight can be toggled by writing 0 and 4 to /sys/class/backlight/<something>/bl_power

I don't know what else to try except looking at the wires with a scope.

klandrith, 2022-01-08:
Ugh, my bad, typo. They switch to out after initialization. I am able to write directly to the backlight pin and toggle it on and off. I got a reply on the Odroid forum from one of the devs behind some software for the Odroid-Go handheld, that uses the same small 2.4" TFT. He stated that the display in that with the init sequence he has will always display white until something is written to it. I presume that the `hktft9340` display type uses the same init sequence and that that is why I am getting the white display.

Now I just need to get a build of ffmpeg or mplayer setup for android so I can test it out. The builds of ffmpeg that I found, for some reason, refused to write directly to the framebuffer and claimed it was an invalid output format.

Thank you again for all the help! I really appreciate it. Have good weekend.

closed 2022-03-18 by github-actions[bot]