notro/fbtft #578: How to use the FT5216 capacitive touch screen of ILI9481

kluoyun, 2022-01-12:
I have the device tree configured and can be displayed.
I don't know if FT5216 touch is supported, if so how can I use it

notro, 2022-01-13:
It's not directly supported by the driver [edt-ft5x06](https://elixir.bootlin.com/linux/latest/source/drivers/input/touchscreen/edt-ft5x06.c) but it looks like many of the focaltech controllers use one of 2 firmwares: https://www.spinics.net/lists/linux-input/msg54001.html, so maybe it will work.

Apparently there's no maintainer for that driver so I suggest you ask on the mainlinglist.
- linux-input@vger.kernel.org
- https://www.kernel.org/doc/html/latest/process/maintainers.html#input-keyboard-mouse-joystick-touchscreen-drivers

kluoyun, 2022-01-14:
alright, thank you very much

closed 2022-01-14 by kluoyun