notro/fbtft #581: radxa zero support?

qNnOtTk, 2022-02-20:
how do I make it work on the radxa zero?
board os: Linux radxa-zero 5.10.69-10-amlogic-g617a45dd0fce aarch64 GNU/Linux Debian 10
SoC is amlogic s905y2
what I tried to get the display running is I used the adafruit18.dts script but modified it a little 
`/*
 * Device Tree overlay for Adafruit 1.8" TFT LCD with ST7735R chip 160x128
 */

/dts-v1/;
/plugin/;

/ {
        compatible = "radxa,zero", "amlogic,g12a";

        fragment@0 {
                target = <&spidev0>;
                __overlay__ {
                        status = "disabled";
                };
        };

        fragment@1 {
                target = <&spi0>;
                __overlay__ {
                        /* needed to avoid dtc warning */
                        #address-cells = <1>;
                        #size-cells = <0>;
                        status = "okay";

                        af18: adafruit18@0 {
                                compatible = "fbtft,adafruit18";
                                reg = <0>;
                                pinctrl-names = "default";
                                spi-max-frequency = <40000000>;
                                rotate = <90>;
                                buswidth = <8>;
                                fps = <50>;
                                height = <130>;
                                width = <130>;
                                reset-gpios = <&gpio_x 8 1>;
                                dc-gpios = <&gpio_x 10 0>; #same thing as below
                                led-gpios = <&gpio_x 9 0>; #here i changed from gpio 25 to gpio_x 9, i referred to the gpio pin names here https://wiki.radxa.com/Zero/hardware/gpio
                                debug = <0>;
                        };
                };
        };

        __overrides__ {
                green = <&af18>, "compatible=fbtft,adafruit18_green";
                speed     = <&af18>,"spi-max-frequency:0";
                rotate    = <&af18>,"rotate:0";
                fps       = <&af18>,"fps:0";
                bgr       = <&af18>,"bgr?";
                debug     = <&af18>,"debug:0";
                dc_pin    = <&af18>,"dc-gpios:4";
                reset_pin = <&af18>,"reset-gpios:4";
                led_pin   = <&af18>,"led-gpios:4";
        };
};

`
Can anyone help me make this display work on this board please?

notro, 2022-02-21:
That compatible won't work, should be ```compatible = "sitronix,st7735r";```

queld, 2022-02-21:
I'll try that when i get home, but that's not the problem,the only parts I changed for this code is compatible = "radxa,zero", "amlogic,g12a"; and reset-gpios = <&gpio_x 8 1>;
                            dc-gpios = <&gpio_x 10 0>; 
                            led-gpios = <&gpio_x 9 0>;
When I get home I can send the code I used on the RPI, the one in the original post is the one I used on the radxa zero

notro, 2022-02-21:
Now I know what those strange compatibles are, rpi linux has patched up the st7735r driver with variants from the now gone fbtft_device: https://github.com/raspberrypi/linux/commit/1b84376666847a19a33e562a3bc25b0e0d259b26

closed 2022-04-20 by qNnOtTk