notro/fbtft #585: fbtft going blank after 2-3 hours

ciobania, 2022-03-27:
Hello notro,

I'm reaching out with 2 problems about the fbtft:

1. it seems like my setup has 2 screens initialized, based on the following logs from `/var/log/messages`:
```
Mar 27 09:35:00 thermopi fbcp[580]: Primary display is 640 x 480
Mar 27 09:35:00 thermopi fbcp[580]: Second display is 320 x 240 16bps
```

When I start my UI, I am not telling it explicitly which `/dev/fb*` to use. However, it defaults on the first one 640x480. If I try to use `/dev/fb1`, the TFT display shows nothing, apart from the fact it's running, is operational.
**Is there a setting to tell it explicitly to use `/dev/fb1`?**

2. After a couple of hours (2-3) of running the UI app, the display goes white. I've tried writing to it directly, via `fbi` command, and nothing changes. The only way to get something displayed again on the screen is to reboot the raspberry.

For me, it seems like the SPI is not communicating anymore with the display/FrameBuffer. Am I wrong in my assumptions, and what would be the cause of that, too much written on the display?
On the other hand, I also believe that point 1 above can have an impact or something to do with what I'm experiencing.

Would you be able to share some wisdom, please? I'm pulling my hair again with this integration.

Thank you kindly

notro, 2022-03-27:
> Is there a setting to tell it explicitly to use /dev/fb1?

That depends on the application.

> After a couple of hours (2-3) of running the UI app, the display goes white. I've tried writing to it directly, via fbi command, and nothing changes. The only way to get something displayed again on the screen is to reboot the raspberry.

If the SPI clock runs too fast this can happen. A transmission error mangles a command and the controller configuration gets messed up. Try unloading the fbtft driver```modprobe -r <driver>``` and then loading it again ```modprobe <driver>``` so the controller configuration is reloaded. If this works try lowering the SPI clock speed.

ciobania, 2022-03-27:
> That depends on the application.
I'm not sure I follow. I use Pygame to drive the UI/display.

My `/boot/config.txt` is as follows:
```
[all]
#dtoverlay=vc4-fkms-v3d
hdmi_force_hotplug=1
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=spi1-3cs
enable_uart=1
dtoverlay=fbtft,spi0-0,rpi-display,bgr,reset_pin=25,dc_pin=24,led_pin=5,rotate=270,debug=3
dtoverlay=ads7846-spi1,penirq=26,xohms=60,pmax=255
hdmi_group=2
hdmi_mode=87
hdmi_cvt 640 480 30 # 6 0 0 0  <-------- IF I set the resolution to 320 240, the first run throws "Segmentation fault", 
                                         possibly because the TFT is not using `/dev/fb1`?
```
My `/boot/cmdline.txt` looks like this:
```
console=serial0,115200 console=tty1 root=PARTUUID=515f9dc9-02 rootfstype=ext4 fsck.repair=yes rootwait fbcon=map:10 fbcon=font:VGA8x8 logo.nologo
```

If I run `modprobe -r fbtft` I get following message:
```
modprobe: FATAL: Module fbtft is in use.
```

Am I missing other configs, or parameters to tell it to use `/dev/fb1` as primary display, rather than `/dev/fb0`?
I don't need X11, nor HDMI on output on this. It's used for a DIY thermostat.
I don't need dual display on it either. I'm surprised it's being so tricky to default to something or configure it.

notro, 2022-03-27:
You have to unload the driver not fbtft, in my case fb_ili9341:
```
$ lsmod
Module                  Size  Used by
fb_ili9341             16384  0
fbtft                  49152  1 fb_ili9341
```

This post indicates that telling pygame what device to use isn't that straightforward:
https://stackoverflow.com/questions/54778105/python-pygame-fails-to-output-to-dev-fb1-on-a-raspberry-pi-tft-screen

Judging from your config.txt (HDMI 640x480) and /var/log/messages ([fbcp](https://github.com/tasanakorn/rpi-fbcp)) I guess that your app writes to /dev/fb0 possibily using dispmanx and then fbcp copies fb0 to fb1 at a fixed interval. This is common use of fbcp for apps that uses the legacy Pi hw accelaration for graphics rendering. I haven't used setups like this myself, not much I can do to help I'm afraid.

Isn't there a forum for this thermostat setup you're using where you can ask someone who knows how it's configured?

ciobania, 2022-03-27:
I mean I'm building my own thermostat, and as such, I don't need/use 2 displays. (I'm referring to 2 displays here, based on the output from logs.

Indeed Pygame starts on `/dev/fb0`, because if I tell it to write over the `/dev/fb1` it does not display anything to `/dev/fb0` which is what the display uses.

Hence I'm asking here.

This is my lsmod. Why does it say my fb_ili9341 is used by **2** devices?

```
fb_ili9341             16384  2
fbtft                  45056  1 fb_ili9341
syscopyarea            16384  1 fbtft
sysfillrect            16384  1 fbtft
sysimgblt              16384  1 fbtft
fb_sys_fops            16384  1 fbtft
backlight              20480  2 fbtft,drm
ads7846                20480  0
```
Can you help me out with what kind of setup you have? I'm not sure how this all works, or if I actually need `fbcp` altogether.
My understanding is that I needed the overlays to be able to display stuff on the TFT.

If I don't need fbcp, how will I get the display to use `/dev/fb1`? I'm not sure how to phrase the question to be more clear of what I'm experiencing. Apologies if I'm a nuisance.

From my understanding, my TFT display is using `/dev/fb0` instead of `/dev/fb1`. Indeed if I use `fbi` to write to `/dev/fb1`, there's nothing displayed on the TFT. If I use `fbi` to write to `dev/fb0`, the image is correctly displayed, and stays there.

However, if I use a loop and write every 0.1 - 1 seconds to `/dev/fb1` then, the TFT display is flickering the photo I'm trying to display. Which sounds pretty much like the `fbcp` is copying from `/dev/fb1` over to `dev/fb0`

Thank you for any suggestions you might have, related to above questions, or clarity in understanding what am I missing from my setup that is so different from yours.

I will try to disable fbcp...see where that gets me.

ciobania, 2022-03-27:
OK, I think now I understand what has happened.
Mixture of different things I've tried over the time, which left my setup in a weird state.
Throwing the `fbcp` in the mix, from previous attempts, did not help with what I was experiencing. No wonder the screen was going white on me, and why the `lsmod` was showing 2 "devices" using it.

I have "disabled" fbcp, from `/etc/rc.local` and now I get the correct behaviour!
Thank you kindly for your time, and explanations.

If you have a paypal/gofundme, I'm more than happy to compensate for your time and expertise!!!

I shall close this now.

closed 2022-03-27 by ciobania