notro/fbtft #586: dmesg errors: no spi_device_id, using dummy regulator

ciobania, 2022-04-01:
Hello Notro,

I am still debugging the issue with the display going full white, after 30 minutes.

Looking in dmesg, I get the following errors:

```
fbtft: module is from the staging directory, the quality is unknown, you have been warned.
fb_ili9341: module is from the staging directory, the quality is unknown, you have been warned.
SPI driver fb_ili9341 has no spi_device_id for ilitek,ili9341
fb_ili9341 spi0.0: fbtft_property_value: buswidth = 8
fb_ili9341 spi0.0: fbtft_property_value: rotate = 270
fb_ili9341 spi0.0: fbtft_property_value: fps = 30
graphics fb1: fb_ili9341 frame buffer, 320x240, 150 KiB video memory, 16 KiB buffer memory, fps=33, spi0.0 at 32 MHz
-----------------------------------------------------------------------
gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
SPI driver ads7846 has no spi_device_id for ti,tsc2046
SPI driver ads7846 has no spi_device_id for ti,ads7843
SPI driver ads7846 has no spi_device_id for ti,ads7845
SPI driver ads7846 has no spi_device_id for ti,ads7873
ads7846 spi1.1: supply vcc not found, using dummy regulator
ads7846 spi1.1: touchscreen, irq 199
input: ADS7846 Touchscreen as /devices/platform/soc/3f215080.spi/spi_master/spi1/spi1.1/input/input0
```

On top of that, screen is rotated 270 degrees, and touch is inverted on 0 degrees (portrait). If I use `swapxy=1` it only turns it around by 90 degrees. Is there a way to force the ads7846 overlay to force it to swap 270 degrees?

notro, 2022-04-01:
The [no spi_device_id](https://github.com/torvalds/linux/commit/5fa6863ba69265cb7e45567d12614790ff26bd56) message is just a warning that the driver should be fixed because module autoloading won't work, but it clearly does so not sure about that one, but it's harmless.

There's no way to invert the axis in this driver, you have to use a calibration tool. Long time ago I used ts_calibrate for this, I don't know if that is still the tool to use.

As for the display going white, this usually means that the configuration has been messed up by a transmission error. Is the display connected through long wires? You can try running the display at 16MHz to see if that is indeed the problem.

ciobania, 2022-04-01:
I used ts_calibrate and it displays fine, I am able to click fine and it draws correctly, it follows the cursor seemingly.

I increased the gpu_mem to 128MB and it run for about 3 hours before displaying again, a white screen.

I was thinking to add some resistors on the data lines or a capacitor on the VCC and GND just in case there's noise somewhere...

If I leave the display sotting on the console, it just sits there for a long time. Is there a programatic way to empty the video memory?

I want to write some sort of test, but I don't know when the display is going white/blank. At which point in time.

If I enable debug=7 my logs will fill the disk.
I wonder how come the RetroPie does not have the same issues?

The wires are standard lenght dupont cables. 5-10cm.