notro/fbtft #587: A request to take a look at this tinydrm patch

t123yh, 2022-04-15:
Hi Notro,

I submitted this patch to kernel mailing list a month ago, but no one is replying me. It's a bug fix to tinydrm SPI, which would cause SPI transfers to be rejected on Rockchip platforms. Could you please take a look?

https://lore.kernel.org/lkml/20220308015611.3007395-1-t123yh.xyz@gmail.com/T/

notro, 2022-04-16:
I had missed that patch, replied now.

closed 2022-04-16 by notro

t123yh, 2022-05-10:
Hi @notro ,

I just sent a modified version of this patch, but your email server thinks my email is a spam.

```
550 5.7.2 Message content rejected (spam)
```

Could you please add a whitelist for my email address?

notro, 2022-05-14:
> Could you please add a whitelist for my email address?

I have no control over the email server, so I can't do anything about it.

Found your patch here:
https://lore.kernel.org/dri-devel/20220510030219.2486687-1-t123yh.xyz@gmail.com/

Looks good at first glance, I'll test it and then apply it when I have some time.