notro/fbtft #588: ILI9341 low fps

N3KRoN, 2022-05-03:
Hello, I am using the ILI9341 screen with the circuit from here on my raspberry pi type B, but I have 2 problems:
1) I can't flip the picture 270 degrees to make it vertical
2) the screen refresh rate is too slow

https://github.com/nopnop2002/Raspberry-ili9325/

https://user-images.githubusercontent.com/11051346/166451879-2f652710-0474-41c0-8b90-2beb5abaf820.mp4

/etc/modprobe.d/fbtft.conf

options fbtft_device name=flexpfb bgr=1 fps=30 gpios=reset:4,dc:3,wr:7,cs:2,db00:15,db01:17,db02:18,db03:27,db04:22,db05:23,db06:24,db07:25
options flexfb width=480 height=320 buswidth=8 init=-1,0x01,-2,5,-1,0x28,-1,0x36,0x58,-1,0x3A,0x55,-1,0x11,-2,100,-1,0x29,-2,20,-3

dmesg:

[    7.703503] graphics fb1: flexfb frame buffer, 480x320, 300 KiB video memory, 4 KiB buffer memory, fps=33
[    7.703661] fbtft_device: GPIOS used by 'flexpfb':
[    7.703676] fbtft_device: 'reset' = GPIO4
[    7.703683] fbtft_device: 'dc' = GPIO3
[    7.703689] fbtft_device: 'wr' = GPIO7
[    7.703694] fbtft_device: 'cs' = GPIO2
[    7.703700] fbtft_device: 'db00' = GPIO15
[    7.703705] fbtft_device: 'db01' = GPIO17
[    7.703711] fbtft_device: 'db02' = GPIO18
[    7.703717] fbtft_device: 'db03' = GPIO27
[    7.703723] fbtft_device: 'db04' = GPIO22
[    7.703728] fbtft_device: 'db05' = GPIO23
[    7.703734] fbtft_device: 'db06' = GPIO24
[    7.703740] fbtft_device: 'db07' = GPIO25
[    7.703809] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[    7.703842] flexfb flexpfb.0: flexpfb id=0 pdata? yes

notro, 2022-05-04:
fbtft is bitbanging the parallel bus making it slow: [fbtft_write_gpio8_wr()](https://elixir.bootlin.com/linux/latest/C/ident/fbtft_write_gpio8_wr).
Using SPI is faster when it can use DMA, but even SPI will be "slow" on a 480x320 display. Some old performance tests: https://github.com/notro/fbtft/wiki/Performance

If you can use SPI and have the Raspberry Pi v5.15 kernel, the [panel-mipi-dbi](https://github.com/notro/panel-mipi-dbi/wiki) driver is a better choice since it can run at higher speeds without messing up the controller configuration.
Ref: [fbtft-vs-drm](https://github.com/notro/panel-mipi-dbi/wiki/fbtft#fbtft-vs-drm)

N3KRoN, 2022-05-04:
Unfortunately, I can't use spi in any way, this screen is an arduino shield with parallel connection... I managed to flip the screen and solve the issue with an update, now the screen shows as it should, but fbtft creates a very heavy load on the rpi type b and works very slowly everything

rsaxvc, 2022-05-12:
Can you use the SMI interface with DMA to drive the parallel bus?

closed 2022-07-20 by github-actions[bot]