notro/fbtft #591: Parallel interface overlay

rbarreiros, 2022-09-04:
Hello,

I'm trying to make an overlay for the ili9486 in 8080 8bit parallel mode, but I'm failing miserably, mainly because I have no idea over what target to use. I made a custom module (ili9486_parallel) to debug and fix any stuff that might arise, but the overlay is boggling me.
Here's what I have right now, I'm guessing the target is not gpio, and it would be nice to have full debug on (which isn't) so I can check out what's happening when I load the overlay, any ideas ?

Thanks.

```
/dts-v1/;
/plugin/;

/ {
        compatible = "brcm,bcm2835";

        fragment@0 {
                   target = <&gpio>;
                   __overlay__ {
                           /* needed to avoid dtc warning */
                           #address-cells = <1>;
                           #size-cells = <0>;
                           ili9486_parallel:ili9486_parallel@0 {
                                        reg = <0>;

                                        compatible = "ilitek,ili9486_parallel";
                                        rotate = <0>;
                                        //bgr;
                                        buswidth = <8>;
                                        reset-gpios = <&gpio 17 0>;
                                        dc-gpios = <&gpio  4 0>;
                                        cs-gpios = <&gpio 27 0>;
                                        wr-gpios = <&gpio 12 0>;
                                        rd-gpios = <&gpio  1 0>;
                                        db-gpios = <&gpio  9 0>,
                                                   <&gpio 11 0>,
                                                   <&gpio 18 0>,
                                                   <&gpio 23 0>,
                                                   <&gpio 24 0>,
                                                   <&gpio 25 0>,
                                                   <&gpio  8 0>,
                                                   <&gpio  7 0>;
                                        /* LED pin drives backlight directly. Use transistor (50mA) */
                                        /* led-gpios = <&gpio 4 1>; */
                                        debug = <1>;
                         };
                };
        };
};

```

notro, 2022-09-05:
You can use `target = <&soc>;`
Nodes under /soc are added as platform devices.

rbarreiros, 2022-09-05:
Thank you very much. It's loading the overlay, but I got some warning even compiling, I'll address these letter. Now on to figuring out how to handle the touch, as it's multiplexed with the LCD .... Any hints on how to proceed?

notro, 2022-09-05:
If they share the same databus, you'll have to write your own touch driver that can coexist with the fbtft driver. Is there a Linux driver for the touch controller?

rbarreiros, 2022-09-05:
I read about someone that managed to have it working, I'm going to inspect it's overlay and see what was done and how, nevertheless, don't think there's a driver.

The LCD is this one

http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO

And by checking the examples for Arduino

http://www.lcdwiki.com/res/Program/Arduino/3.5inch/UNO_8BIT_ILI9486_MAR3501_V1.1/3.5inch_Arduino_8BIT_Module_ILI9486_MAR3501_V1.1.zip

I can see it's multiplexing the touch with the LCD (check touch demo)

Who I read that has this working (it used the retired flexfb) is this:

https://github.com/goodtft/LCD-show

Check file LCD35-show (shell script), later I'll inspect the overlay used in that script.

notro, 2022-09-06:
AFAICT there's no touch controller on that display, the microcontroller acts as the touch controller using its ADC. The Raspberry Pi doesn't have an ADC so you would need a touch controller of some kind.

rbarreiros, 2022-09-11:
I managed to get it working decently, using the ILI9486 module, and the overlay below, no need to write a new module. If there's tweaking to be made on the ini, I think it could be done in the overlay, it would replace the init in the module (am I thinking right?)

The touch won't work on a regular Pi, as was said, the Pi's don't have an analog input, and this module without a touch controller requires 2 analogue inputs which are being multiplexed with the LCD pins, so this display on a Pi, it'll be only display. On another SBC it might be possible, but would require a new module perhaps, as it would need to keep polling the touch for new events and send them to the event system.

Anyway, here's the overlay if there's any interest in adding it to the list of overlays available

```
/dts-v1/;
/plugin/;

/ {
        compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709", "brcm,bcm2710";

        fragment@0 {
                   target = <&soc>;
                   __overlay__ {
                           ili9486 {
                                        compatible = "ilitek,ili9486";
                                        bgr;
                                        fps = <25>;
                                        width = <320>;
                                        height = <480>;
                                        rotate = <90>;
                                        buswidth = <8>;
                                        reset-gpios = <&gpio 17 1>;
                                        dc-gpios = <&gpio  4 0>;
                                        cs-gpios = <&gpio 27 1>;
                                        wr-gpios = <&gpio 12 1>;
                                        rd-gpios = <&gpio  1 1>;
                                        db-gpios = <&gpio  9 0>,
                                                   <&gpio 11 0>,
                                                   <&gpio 18 0>,
                                                   <&gpio 23 0>,
                                                   <&gpio 24 0>,
                                                   <&gpio 25 0>,
                                                   <&gpio  8 0>,
                                                   <&gpio  7 0>;
                                        debug = <0>;
                         };
                };
        };
};

```

closed 2022-11-19 by github-actions[bot]