notro/fbtft #593: The 6.0 kernel uses tbftf and there is a problem that fbtft cannot be mounted

kluoyun, 2022-12-30:
As shown below, the log after I execute "modprobe fbtft" and "modprobe fb_ili9340" is as follows, debug=7 in the device tree, spi node and tft are also configured correctly.

```
root@x:~# dmesg | grep fb
[   28.602048] fbcon: Taking over console
[   67.602107] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   74.034889] fb_ili9340: module is from the staging directory, the quality is unknown, you have been warned.
[   74.045035] SPI driver fb_ili9340 has no spi_device_id for ilitek,ili9340
root@x:~# 

```

notro, 2023-01-01:
Commit [5fa6863ba692](https://github.com/torvalds/linux/commit/5fa6863ba69265cb7e45567d12614790ff26bd56) ("spi: Check we have a spi_device_id for each DT compatible") added that warning.

The [FBTFT_REGISTER_DRIVER](https://elixir.bootlin.com/linux/latest/C/ident/FBTFT_REGISTER_DRIVER) macro will need to be fixed somehow to get rid of the warning.

kluoyun, 2023-01-06:
Thanks for your reply, I will try

closed 2023-01-06 by kluoyun

notro, 2023-01-06:
It's just a warning so the driver should work as-is, but if you find a fix it's all the better.

kluoyun, 2023-01-06:
I modified the SPI driver and device tree and it works