notro/fbtft #594: support for ssd_1305

anshulmakkar, 2023-03-14:
I couldn't find the driver for ssd1306. Has its support been dropped ?

notro, 2023-03-17:
It's in mainline Linux: https://elixir.bootlin.com/linux/latest/source/drivers/staging/fbtft/fb_ssd1305.c

tomsn1978, 2023-03-20:
Hello I want to initialize the screen to black when I boot the kernel，Where should I modify the source code

anshulmakkar, 2023-03-22:
got you. thanks..
Won't it be useful to have the following initialization for CS and DS line in the init function:

+++ b/drivers/staging/fbtft/fb_ssd1305.c
@@ -43,6 +43,8 @@ static int init_display(struct fbtft_par *par)
                mutex_unlock(&par->gamma.lock);
        }

+       par->gpio.reset = 1;
+       par->gpio.dc = 0;
        /* Set Display OFF */
        write_reg(par, 0xAE);

notro, 2023-03-26:
'reset' and 'dc' are pointers to [struct gpio_desc](https://elixir.bootlin.com/linux/latest/C/ident/gpio_desc) so this won't work ([struct fbtft_par](https://elixir.bootlin.com/linux/latest/C/ident/fbtft_par)).

Why do you want to do this?

closed 2023-06-03 by github-actions[bot]

anshulmakkar, 2023-06-09:
The issue can be closed.