notro/fbtft #597: IlI9431 white screen with F1C100S (lichee nano)

mzashh, 2023-07-12:
so i am trying to get an ili9341 working with the lichee nano. some people have already done it so ik that it is possible but all im getting is a white screen, my display module has a CS pin and is working fine with the Adafruit ILI9341 library, fbtft loads fine but the display is still blank, i used this guide 
https://lualiliu.github.io/website/f1c100s-fbtft.html

here are the dts and dtsi, i have the display set up under spi0
[suniv-f1c500s-miyoo-4bit-1.dts.txt](https://github.com/notro/fbtft/files/12032778/suniv-f1c500s-miyoo-4bit-1.dts.txt)
[suniv.dtsi.txt](https://github.com/notro/fbtft/files/12032782/suniv.dtsi.txt)

also here is the dmesg
[dmesg-1.txt](https://github.com/notro/fbtft/files/12032792/dmesg-1.txt)

notro, 2023-07-14:
```
				reset-gpios = <&pio 4 11 GPIO_ACTIVE_LOW>;
				dc-gpios = <&pio 4 10 GPIO_ACTIVE_LOW>;
```
dc-gpios should be GPIO_ACTIVE_HIGH. The D/CX pin on the controller should be low when the command byte is sent and the driver does this by [setting the gpio to zero](https://elixir.bootlin.com/linux/v4.14/source/drivers/staging/fbtft/fbtft-bus.c#L40).
reset-gpios changed behaviour in some kernel version, but on [4.14](https://elixir.bootlin.com/linux/v4.14/source/drivers/staging/fbtft/fbtft-core.c#L343) it should be GPIO_ACTIVE_HIGH.

mzashh, 2023-07-15:
notro thanks for the reply, i did change it but it seems to be making no difference, the screen is still white

notro, 2023-07-15:
A white display means that it's not been configured correctly.
I assume you've verified that the init sequence in dmesg is the same as the adafruit library.
You can try to lower the speed in case there's noise on the line.
Apart from that it's gpio configuration or wiring issues that can break things.

mzashh, 2023-07-16:
The init code is the same as UCGlib, and my display works fine with that library as well.
The wiring is ok i have even tried soldering it directly.
I tried lowering the speed still nothing.
my only guess is that the gpio config might be messing something, up is there a particular command to like see what pins fbtft is using like other than the reset and DC ones or anything else i can try to verify?

notro, 2023-07-16:
If your platform has a pinctrl driver, you can see the config under `/sys/kernel/debug/pinctrl`.
The Raspberry Pi has a tool `raspi-gpio` that can dump the config directly from the registers, I don't know if something like that exists for your device.

mzashh, 2023-07-18:
the pinctrl seems to be ok as well, i am using spi0 and using the correct pins

	SUNXI_PIN(SUNXI_PINCTRL_PIN(C, 0),
		  SUNXI_FUNCTION(0x1, "gpio_out"),
		  SUNXI_FUNCTION(0x2, "spi0"),		/* CLK */
		  SUNXI_FUNCTION(0x3, "mmc1")),		/* CLK */
	SUNXI_PIN(SUNXI_PINCTRL_PIN(C, 1),
		  SUNXI_FUNCTION(0x0, "gpio_in"),
		  SUNXI_FUNCTION(0x1, "gpio_out"),
		  SUNXI_FUNCTION(0x2, "spi0"),		/* CS */
		  SUNXI_FUNCTION(0x3, "mmc1")),		/* CMD */
	SUNXI_PIN(SUNXI_PINCTRL_PIN(C, 2),
		  SUNXI_FUNCTION(0x0, "gpio_in"),
		  SUNXI_FUNCTION(0x1, "gpio_out"),
		  SUNXI_FUNCTION(0x2, "spi0"),		/* MISO */
		  SUNXI_FUNCTION(0x3, "mmc1")),		/* D0 */
	SUNXI_PIN(SUNXI_PINCTRL_PIN(C, 3),
		  SUNXI_FUNCTION(0x0, "gpio_in"),
		  SUNXI_FUNCTION(0x1, "gpio_out"),
		  SUNXI_FUNCTION(0x2, "spi0"),		/* MOSI */
		  SUNXI_FUNCTION(0x3, "uart0")),	/* TX */

so now i am really confused as to why its working

notro, 2023-07-25:
You can try a spi loopback test: https://elinux.org/RPi_SPI#Loopback_test

closed 2023-10-02 by github-actions[bot]