notro/fbtft #599: Problems about the mainline linux and this repo

gtxzsxxk, 2023-08-30:
Are you still maintaining the mainline linux's fbtft drivers? I see that there are quite a few differences between the mainline linux and this repo. Take a probable mistake in the mainline `fbtft-core.c` for example.

mainline code
```c
static void fbtft_reset(struct fbtft_par *par)
{
	if (!par->gpio.reset)
		return;

	fbtft_par_dbg(DEBUG_RESET, par, "%s()\n", __func__);

	gpiod_set_value_cansleep(par->gpio.reset, 1);
	usleep_range(20, 40);
	gpiod_set_value_cansleep(par->gpio.reset, 0);
	msleep(120);

	gpiod_set_value_cansleep(par->gpio.cs, 1);  /* Activate chip */
}
```

this repo
```c
void fbtft_reset(struct fbtft_par *par)
{
	if (par->gpio.reset == -1)
		return;
	fbtft_par_dbg(DEBUG_RESET, par, "%s()\n", __func__);
	gpio_set_value(par->gpio.reset, 0);
	udelay(20);
	gpio_set_value(par->gpio.reset, 1);
	mdelay(120);
}
```

Note that for most tft-lcds the reset logic is negative, i.e. when the reset pin is low-level, the lcd will reset. And the mainline code is causing problems. The comment is `activate chip` but what is it doing? It is resetting the lcds because it keeps the reset pin low and it is disabling the lcds too because it keeps cs high. I believe the mainline code is having some typo because when I make the `par->gpio.cs` into `par->gpio.reset` every thing begins to work.

notro, 2023-08-30:
Maintainer status from the frontpage wiki:
`
fbtft is currently listed as orphaned in MAINTANERS so it doesn't have someone looking after it, at least officially.
`

Status of the code in this repo: https://github.com/notro/fbtft/blob/master/README#L4-L6

Unfortunately the reset polarity was changed in this commit: https://github.com/torvalds/linux/commit/ec03c2104365ead0a33627c05e685093eed3eaef

the `cs` gpio is used for chip select on a parallell bus.

gtxzsxxk, 2023-08-30:
Your answer explains. Thanks.

closed 2023-08-30 by gtxzsxxk