notro/fbtft #602: Raspberry Pi 4

NickDG, 2023-11-14:
Hi Notro,

I have a raspberry Pi 2GB and purchased the following screen https://www.amazon.de/-/en/dp/B08TCFDX8B?psc=1&ref=ppx_yo2ov_dt_b_product_details , thinking it would be easy to connect to the raspberry PI, I see its nothing easy. Iv wired up the Pi to the screen as mentioned in the Generic 7" screen running a ARM SSD1963, but not I'm lost, I'm not sure if I need to run a specific version of raspberry OS, and I'm confused on what to run to get this screen working. If you could assist it would be greatly appreciated.

NickDG, 2023-11-14:
All i get when i turn on the Pi is a white screen nothing else

notro, 2023-11-18:
You need a Device Tree overlay that describes the display. I haven't seen a working overlay for a parallel bus display. This is what I wrote in 2018: https://forums.raspberrypi.com/viewtopic.php?t=208264#p1288018

Here is someone that have at least looked at the problem for a SSD1963 #467 

Be aware that there has been some changes to gpio polarity handling in Linux v5.14: [staging: fbtft: Rectify GPIO handling](https://github.com/torvalds/linux/commit/ec03c2104365ead0a33627c05e685093eed3eaef)

jamesc2151, 2023-11-20:
You code help me jump thru many hoops and directions but just want to post a Thanks and update of what I got working using Framebuffers and this might help you somehow..

Working with Raspberry 64bit last release before bookworm  I finally got an generic  ili9341 SPI tft display too correctly show me a booting console screen as it booted... still tweaking it but so far here is what I had to do

Etc\modules load --- spi_bcm2835
Config text.........
hdmi_group=2
hdmi_mode=4
hdmi_mode=4
hdmi_cvt 480 360 60 6 0 0 0
hdmi_drive=2
enable_uart=1
dtoverlay=fbtft,spi0-0,ili9341,reset_pin=24,dc_pin=25,led_pin=18,rotate=90

The overlay is already part of the OS i just had to find the right settting..
The display is https://www.amazon.com/dp/B09XHJ9KRX?ref=ppx_yo2ov_dt_b_product_details&th=1

I don't have the touch connected but will be playing with it latter on...  I did note that launching kodi from the tft gave me kodi output on the Primary HDMI and left the TFT showing background stats of errors it thows up while running so now I know why kodi freezes on me is because it's trying to reload MPD which I have running already..

Thanks for the info and keep on keep'n on
JC

closed 2024-01-27 by github-actions[bot]